alexforencich/verilog-ethernet #4: eth_crc_8.v missing

kabaka0, 2017-11-05:
There are many references to module eth_crc_8 (..64), but I can't find the corresponding source files in the repository? Did you remove them?

alexforencich, 2017-11-05:
Where are these references to eth_crc_*?  Those files were replaced some time ago and all module references should now point to lfsr.v.

kabaka0, 2017-11-05:
As an example,  verilog-ethernet/rtl/axis_eth_fcs_64.v. I was hoping to directly use some of your internal elements (like axis_eth_fcs_64).

alexforencich, 2017-11-06:
That file doesn't reference eth_crc_*?  All those are instances of lfsr.

kabaka0, 2017-11-06:
My mistake, I read through it really first and assumed there was a semi-colon before eth_crc_8 which there clearly isn't!. Thanks for the prompt reply though, and nice job with this repo!

closed 2017-11-06 by kabaka0