alexforencich/verilog-ethernet #5: ML605 example

themperek, 2018-02-24:
According to https://www.xilinx.com/products/boards-and-kits/ek-v6-ml605-g.html ML605 is XC6VLX240T but the example specifies xc6vlx130t (https://github.com/alexforencich/verilog-ethernet/blob/master/example/ML605/fpga_sgmii/fpga/Makefile).

Is there a particular reason for this?
I can provide PR if this is a mistake.

alexforencich, 2018-02-24:
That's what the documentation would indicate. However, the ML605 that I have was apparently stuffed with a 130T. So it would seem that at least some subset of ML605 boards are populated with 130T parts instead of 240T. I'll think about how to properly address this, maybe a Makefile for each part.

alexforencich, 2018-02-26:
This should be fixed now; I have separate makefiles in separate subdirectories for the 130t and 240t versions.

closed 2018-02-26 by alexforencich