alexforencich/verilog-ethernet #7: First byte of Ethernet preamble being replaced by XGMII start byte

philn16, 2019-02-12:
Looking at **test_axis_xgmii_rx_64.py**, and testing **axis_xgmii_rx_64.v** it seems that current behavior ignores the first 8 XGMII bytes (including the start byte 0xFB). However, I'd expect it to ignore the first 9 XGMII bytes (including the start byte 0xFB) since the ethernet preamble + start of frame delimiter is 8 bytes.
Additionally, in **xgmii_ep.py** (which is used to test the XGMII Rx) the first byte is replaced by the XGMII start byte. That is, one of the preamble 0x55 bytes isn't sent through the XGMII.
```python
dl[0] = XGMII_START
cl[0] = 1
```
I added a print out for d and c in  **xgmii_ep.py** when running **test_axis_xgmii_rx_64.py** and got what's below. You can see 0x55 is only repeated 6 times instead of 7.
```
d is d5555555555555fb
c is 1
```

closed 2019-02-12 by philn16

alexforencich, 2019-02-12:
The XGMII start character replaces the first preamble byte as per IEEE 802.3.

alexforencich, 2019-02-12:
Specifically IEEE 802.3 46.2.2.