alexforencich/verilog-ethernet #9: How to apply your project into ZC706?

JackieLee0524, 2019-03-20:
Dear author:
    First,thanks for your sharing the code,but now i am going apply it on ZC706.Of course i have found some data,then i find that ZC706 can only use the SFP.so  i wonder if i can use your code or which part i can use.i am a beginning student,but i want to learn.so can you tell me how to put your code on the ZC706.Thank you very much.

alexforencich, 2019-03-20:
I see, the RJ-45 jack on that board seems like it is only accessible via the ARM core, so you have to use the SFP+ connector.  There should be a few different options for using that connector.  If you want to run at 1 Gbps, then use the 1G/2.5G PCS/PMA core and edit one of the gigabit example designs, probably the VCU108 would be the best one to look at.  If you want to run at 10G, then use the 10G/25G PCS/PMA core and take a look at the VCU108 example design, or use the included PHY with a transceiver wizard instance based off of the VCU118 example design.

JackieLee0524, 2019-03-20:
ok,thanks again.your reply is very useful.I just want run at 1Gbps,and i will follow your reply to make further study.thank you very very much.

jreindel, 2019-07-18:
I am using the MAC and PHY core to run a 10Gb Ethernet through the Xilinx GTH transceiver core set to 10G-BASER. Check out the 7000 series gtwizard product guide:
 https://www.xilinx.com/support/documentation/ip_documentation/gtwizard_ultrascale/v1_7/pg182-gtwizard-ultrascale.pdf 
and the 7000 series gtwizard user guide: https://www.xilinx.com/support/documentation/ip_documentation/gtwizard/v3_6/pg168-gtwizard.pdf from Xilinx.
Sounds like one of the GTX transceivers is connected to the SFP+ on the ZC706.

alexforencich, 2019-07-18:
For an SFP module, the standard you want is 1000BASE-X.  I don't have a 1G PCS/PMA core in this repo, so it will be necessary to use the (free) 1G PCS/PMA core from Xilinx.  And that should take care of the transceiver configuration for you, all you need to do is tell it what kind of transceiver you want to use.  Should be this guide: https://www.xilinx.com/support/documentation/ip_documentation/gig_ethernet_pcs_pma/v16_0/pg047-gig-eth-pcs-pma.pdf