alexforencich/verilog-ethernet #10: Any documents on how to use UDP stack?

sinaaghli, 2019-04-02:
Thanks for sharing this amazing library, this pretty much has everything I need. 
I consider myself a beginner in HDL programming, and I find it very hard to use this library. 
I need to implement a 1G UDP on zedboard (which has a 1g Ethernet FMC card) and I was trying to mimic the project style you have in your examples folder but I wasn't sure how to compile the project or where to make modifications. Your make files don't seem to be a gcc make. 
would you be able to give some advice on how to use this library,

alexforencich, 2019-04-02:
well, it's definitely not going to work with GCC, that's for sure.  If you're not on a system that supports symlinks (i.e. linux or WSL) then you'll probably have to forego the makefiles and pull everything into vivado manually.  If you are on a system that supports symlinks, then you should be able to just run 'make' after sourcing vivado's settings.sh file.

sinaaghli, 2019-04-02:
Thanks for quick response, I really appreciate it.
I would definitely try that but for now I'll try to pull separate files into vivado. 
For that I have pulled udp_complete.v and its dependencies as well as eth_mac_1g_rgmii_fifo.v with its dependencies
The way I understood is that I need to do the following,

1. connect AXI bus on udp_complete to AXI bus on eth_mac_1g_rgmii_fifo
2. connect my application layer buffers to udp_complete interface
3. connect rgmii pins in eth_mac_1g_rgmii_fifo to corresponding fpga pins

After this point I should be able to send data over network, is that correct? am I missing something?
they are many pins on udp_complete that I'm trying to understand what they do but do you mind if I ask occasional questions  if I couldn't figure out?

alexforencich, 2019-04-02:
Yep, that's pretty much right.  You'll probably also want to pull in the proper TCL timing constraints files.  Take a look at the example designs for what pins to connect on udp_complete and what to do with any unused interfaces.  Since the UDP stack has both IP and UDP layers, there are some IP related ports brought out that are optional if you're only using UDP.

sinaaghli, 2019-04-03:
Thanks for the feedback, I wasn't aware of TCL files. I'm working on transferring ML605 example with UDP loopback. It has been going smoothly so far. 
Again, thanks for sharing this amazing library.

alexforencich, 2019-04-03:
Ah, I guess you didn't specify what part and what toolchain you're using.  The TCL timing constraints currently apply only to Vivado, I need to look in to how to something similar for ISE and Quartus.

sinaaghli, 2019-04-04:
The part is a Zynq 7000, I'm using vivado and I pulled in eth_mac_1g_rgmii.tcl as you suggested.
I was able to transfer the library .v files from to a vivado project while following your ML605 example, everything looks fine other than a synthesis error. it says

> [Common 17-55] 'get_property' expects at least one object. ["eth_mac_1g_rgmii.tcl":23]

do I need to pull other tcl files too? I basically added the tcl file as a constraint file to the project, I thought thats the way to go (sorry I have no experience with tcl files. I need to study on what tcl file does, once I get this running).

alexforencich, 2019-04-04:
A better example design to look at is the one for Nexys Video.  The makefile there should list all of the required TCL files.  You should be including rgmii_phy_if.tcl (timing constraints for the reset logic and clk90 in the PHY interface), eth_mac_1g_rgmii.tcl (timing constraints for the speed sense logic), eth_mac_fifo.tcl (timing constraints for the status synchronization), and axis_async_fifo.tcl (timing constraints for the asynchronous FIFOs).  And that's exactly what you do, you add it as a constraint file, you'll have to change the drop-down in the file browse dialog from XDC to TCL to locate it.  You should not be getting any errors from the TCL scripts - they should be set up to deal with things that can get optimized out.  If it can't find something, then something may be configured or connected incorrectly.

sinaaghli, 2019-04-06:
Thanks for the detailed explanation. After adding all those tcl files and tripple checking the project I wasn't still able to pass Synthesis. 
So i took your advice and started a new project and worked on converting Nexyx one this time. This one only had didn't use any tcl files so I let that be so. I am able to synthesize and also implement the project now, which is very exciting. 
there are some issues with FMC cards jtag connection which I'll fix tomorrow but other than than I have one question for you,

- ethernet rgmii ports in your design has two input ports calle phy_int_n and phy_pme_n but my phy chip doesn't use those. instead I have two extra pins on my phy chip called ETH0_MDC and ETH0MDIO I don't think these two correlate to each other but do you have any advice for me. 
I thought rgmii is standard, I'm not sure why there are differences between different chips. 
P.S. my fmc card(avnet network FMC) uses KSZ9031RNXIC chip

sinaaghli, 2019-04-06:
I just realized phy_int_n and phy_pme_n are not connected to anywhere in fpga_core so nvm that question. 
Now I wonder if I need to do sth with MDC and MDIO pins in my FMC card.

alexforencich, 2019-04-06:
"not being able to pass synthesis" implies some sort of an error, it could be useful to see what the error is.  Actually, it's possible that you ran in to an issue with the rgmii_phy_if parameters - they have to be set correctly for different parts so the right clocking and IO primitives are used, and I think these settings are differerent between 6 series and 7 series and would cause errors if set incorrectly.  

For the Nexys Video example design, the TCL files you need should be listed here: https://github.com/alexforencich/verilog-ethernet/blob/master/example/NexysVideo/fpga/fpga/Makefile#L49 .  Without these, you should be able implement the design, but the timing either will not be correctly constrained or it will fail.  

The int_n and pme_n signals are not used at the moment.  The mdc and mdio signals may or may not be required, depending on the PHY chip.  If it comes out of reset with sensible defaults, then you can leave mdc and mdio floating.  If it does not, then you might need to do some register writes to get things running.  The SGMII PHY chip from TI on the VCU118 board, for example, does not come up with sensible defaults and hence requires a number of register writes in order to function correctly: https://github.com/alexforencich/verilog-ethernet/blob/master/example/VCU118/fpga_1g/rtl/fpga.v#L371 .

sinaaghli, 2019-04-10:
Short update, after spending lots of time on making my vivado project work I could not pass the errors for 

>`[Common 17-55] 'get_property' expects at least one object. ["eth_mac_1g_rgmii.tcl":23]`

which was also repeated for all other tcl files. So I gave the compiling option on linux another try and that worked like a charm.  after running make I was able to open the compiled project file using vivado and modify ports to zedboard. 
For some reason the FMC card breaks my jtag chain which appears to be a known hardware issue on FMC card which I'm trying to debug at the moment. I'll let you know how it goes once I was able to program the fpga.

thanks again for all your helpful comments.

IncognitoChap, 2021-11-08:
I too faced the tcl file error, and the way I resolve it was to readjust the sequence of constraints(by putting the fpga.xdc first). I am not sure if this is the right way to resolve it. Is the order at which Vivado interpret the constraint files something worth to investigate?

closed 2022-08-26 by sinaaghli