alexforencich/verilog-ethernet #11: udp_complete for tx only

olagrottvik, 2019-04-03:
Thanks for this IP!

However, I have a small issue. I'm trying to figure out exactly what signals are needed to produce just a single UDP TX packet, but I can't seem to produce anything else than ARP packets. I'm trying to use the test_udp_complete.py as a guide. I noticed that if I commented out all other test cases but testcase5(UDP TX), the test failed. It only works if I uncomment testcase2(IP TX). When testcase2 is commented out, it seems that I'm only creating an ARP packet (type= 0x0806) and not IPv4 (type = 0x0800). Another hint for what I'm doing wrong might be that udp_tx_busy goes high, and stays high forever.

What exactly is it that happens in testcase 2 that is required to create a UDP TX packet in testcase5?

Any help is greatly appreciated!

alexforencich, 2019-04-03:
Test 2 also includes a response to an ARP request from the stack.  If the IP layer doesn't receive an ARP response to the address you're trying to send do, you won't be able to send any data.  For any top-level testbench that uses the UDP stack, you'll have to do something to get the IP/MAC combo for all destination IP addresses you want to use into the ARP table, either by handing the ARP requests as the core generates them or by sending gratuitous ARP requests that the core will use to populate the ARP table.  

Addresses in incoming ARP packets are stored in the ARP table here: https://github.com/alexforencich/verilog-ethernet/blob/master/rtl/arp.v#L273

alexforencich, 2019-04-03:
aaaand I just realized that the wrong constant is used for inarp requests, whoops

alexforencich, 2019-04-04:
Oh, I guess I should add that if you want literally TX only with no RX at all, then you'll have to just chain together the UDP TX and IP TX modules without any of the extra 'glue' that's included in ip_complete and udp_complete.  If you do that, then you'll have to supply the proper destination MAC address in addition to the IP address.

olagrottvik, 2019-04-04:
Ah, that makes sense. Thanks for the ultra-fast reply!

olagrottvik, 2019-04-04:
Hi, sorry for reopening this, but just have a quick follow up question: 
Is there a way to send a Gratuitous ARP Request with udp_complete? If I e.g. wanted to announce the mac-address and the IP to the network on boot-up.
If not, maybe it would be a nice feature to add?

alexforencich, 2019-04-04:
That's a good idea for a feature; I'll look in to adding some ports to request sending one.  It would be pretty simple in the meantime to add another ARP TX module and use an eth_arb_mux module to insert it into the transmit path.

olagrottvik, 2019-04-04:
Neat! Thanks

dgarigali, 2021-01-18:
> That's a good idea for a feature; I'll look in to adding some ports to request sending one. It would be pretty simple in the meantime to add another ARP TX module and use an eth_arb_mux module to insert it into the transmit path.

Hi, Alex, this feature has been developed already?