alexforencich/verilog-ethernet #12: Unconnected clear_cache in arp_cache.v

olagrottvik, 2019-05-28:
Hi Alex,

Once again, thanks again for all your hard work. I've started to make out how to use most of your stuff now. However, the clear_cache wire on the arp_cache.v module seems to be unconnected. Any plans for adding some clear functionality in the near future?

Ola

alexforencich, 2019-05-28:
It's a simple enough pin to pull out to the top level.  I suppose I can go ahead and do that.  Is there a particular reason you need to clear the ARP cache?

olagrottvik, 2019-05-28:
No, what I meant is that it does not seem to do anything inside of arp_cache.v, so I was wondering why the pin is there at all when it doesn't do anything. I might be missing something obvious? I guess you added the pin to arp_cache.v at some point for it to actually be possible to delete cache. Also, the pin is already routed to the top-level, however, it has no effect of pulling it high.

It's not a critical feature for me, just curious as the pin is actually there. Also, actually pulling "rst"-pin high does not seem to clear the cache either?

alexforencich, 2019-05-28:
Ah, right.  That makes more sense.  The ARP cache was rewritten at one point to make it more efficient (i.e. use a RAM instead of a bunch of registers), and I guess I forgot to add that functionality to the new version.  I'll take a look at adding that back when I get a chance.  Thanks for the heads up!

olagrottvik, 2019-05-28:
Cool, and thanks again!

alexforencich, 2019-05-28:
Actually, taking a look at the code this should be super simple and would just push the valid bit into flip flops, which is not too bad.  I will get that fixed shortly.

alexforencich, 2019-06-14:
This is implemented now.  I did not push the valid bit into flip flops as that would have added ~512 flip flops in the default configuration (512 entries), however the cache clear operation will take one clock cycle for each cache entry, which is a reasonable trade-off given the low implementation overhead.

closed 2019-06-14 by alexforencich