alexforencich/verilog-ethernet #13: Use of tkeep in the middle of a packet for ignoring bytes

olagrottvik, 2019-06-26:
Hi Alex,

Sorry for all the questions today. I have a small issue with using tkeep with eth_mac_10g_fifo. It seems like tkeep is not really a byte qualifier if it is used somewhere in the stream before tlast. It will only remove the last bytes in the stream.

For instance, this is an input:
[image]

The bytes that are in the position of the unasserted tkeep will be kept in the resulting xgmii output. However, the last four bytes of the stream are dropped, even though tkeep is asserted for all of them. This issue seems to originate from axis_xgmii_tx_64.v, where non tkeep bytes are really only dropped if they are in the last window. 

My question is this: is this intended behavior and do the IP in general require that a complete stream with no gaps are delivered?

alexforencich, 2019-06-26:
The MAC only looks at tkeep when tlast is high.  It's not really possible to change that as the data must be contiguous on the XGMII interface,, hence why the MAC is usually used with a frame mode FIFO that will not release data until it has a whole frame.  The MAC logic is also partially simplified as it essentially feeds tkeep through a priority encoder - you can't drop arbitrary bytes, only some number from the end of the frame.  However, it should be masking bytes to 0 that have the corresponding tkeep signal low, no matter where they are in the packet.  It would be possible to write a module to remove those bubbles before the FIFO, but this would require quite a bit of tedious logic to store partial words, barrel-shift things into alignment, etc.  

The intention of tkeep from the point of the MAC module is to enable transmission of frames that are not a multiple of 8 bytes in length.  The MAC expects the packet data to be contiguous, as it has to be contiguous on the XGMII interface - it will generate XGMII error control characters and drop the remainder of the frame if tvalid falls within a packet, for example.  So it really only implements a subset of the possible tkeep functionality, and operation outside of that is undefined.

alexforencich, 2019-06-26:
I just took a peek at the AXI lite spec, and it says that master and slave devices are not required to support 'null' bytes (bytes marked with tkeep low).  It also states that for devices that do not support 'null' bytes, the interconnect must remove them (called 'packing').

olagrottvik, 2019-06-26:
Thanks for your quick reply. I'll adopt my design to adhere to those requirements.

closed 2019-06-26 by olagrottvik