alexforencich/verilog-ethernet #17: Problems trying to combine GMII MAC Module with Xilinx PCS/PMA IP

KevinSmild, 2019-08-27:
Hello Alex, thanks for this amazing repository.

I'm having some troubles trying to connect your modules with the PCS/PMA IP. I started from the VCU108 example (which is also using the IP but with a LVDS transceiver) and made some changes. I'm using a device specific transceiver for 1G 1000BASE-X, so I changed the IP accordingly. I also changed to the eth_mac_1g_gmii_fifo module in the fpga_core file. I'm pretty sure that the PCS/PMA IP is well configured by checking at the last bit in the pcs_pma_status_vector. However, I cannot receive any packets from the FPGA.

[image]
[image]
[image]

I am not sure what I'm doing wrong.

KevinSmild, 2019-08-27:
Thanks for the reply. 
Yes, it sythesized without any errors. I tried what you just suggested at the beginning and it didn't work. That's why I had the idea to change to the eth_mac_1g_gmii_fifo module (which now I know is wrong).

KevinSmild, 2019-08-28:
Hi, Alex. Again, thanks for your reply.

About the code, essentially is exactly the same code as in the example for the VCU board, with the IP changed to a device specific transceiver and added another clock at 200MHz for the transceiver's independent input clock. 

As you suggested, I tried looking at the RX and TX connection between your module and the PCS/PMA IP. When looking at the RX I get the frames BC 42 00 00 and BC B5 00 00 when no cable is connected, which I guess is correct. However, when I connect the cable I only get the idle frame BC 50.

alexforencich, 2019-08-28:
The MAC ignores the data from the PHY unless rx_dv is high. Does that signal ever go high? If not, are you sure the PHY is set up correctly? Are you sure packets are actually being sent down the link from the other end?

KevinSmild, 2019-08-28:
Thanks for your help, Alex.
I made a silly mistake with the clocking logic. There was a small problem between the clock I gave to your module and the phy_gmii_clk I used. Now everything is working fine. Thank you so much.

I also wanted to ask, have you done any performance measurements for your module?

Again, thank you so much Alex. Your module is amazing, keep up the good work.

alexforencich, 2019-08-29:
Those silly mistakes can take and annoyingly long time to figure out!  On one of my more recent projects, I accidentally mixed up the receive clocks from the transceivers (I connected the four receive clocks for QSFP0 to the PHYs for QSFP1, and vise-versa).  Interestingly, the frequencies were close enough that I could actually send some packets over the link, albeit with very high packet loss.  Didn't realize what happened until I looked at it with the ILA and saw idles....garbage....idles....garbage...

I have not done much in terms of performance measurements.  This particular design is simple enough that it is completely deterministic and most of the behavior can be inferred from the architecture directly.  For example, there is nothing in the design that precludes it from running at line rate at any frame size, aside from perhaps blocking while waiting for an ARP request.

closed 2020-03-04 by alexforencich