alexforencich/verilog-ethernet #19: gig_ethernet_pcs_pma_0 IP on board VCU108 (1g example)

ranjanbl, 2020-01-31:
Hi Alex,
First of all, want to thank you for such an amazing project. I am new to FPGA development and currently using Vivado 2019.2 version with VCU108 board. I am trying the 1g example.

1st Attempt  : Linux env. -->
 I followed your instruction to the word and found the following issue 
"WARNING: [Vivado 12-4802] The synthesis checkpoint for IP '../gig_ethernet_pcs_pma_0.xci' is not generated and IP is locked. An out-of-context (OOC) run will be created and/or launched, but synthesis may not be able to complete or could result in incorrect behavior."
I believe the IP is upgraded (found in the IP catalog) and not synthesizing correctly. This caused implementation to fail.

2nd Attempt : Windows env. -->
I manually pulled every code for the experiment, IP, .xdc constraints, .tcl constraints. Upgraded the IP to the newer version present in IP catalog (GUI has option to upgrade IP). This completed synthesis, implementation, generated bit-stream and programmed the device. But didn't see the result as expected in netcat. No returned characters after pressing enter.

alexforencich, 2020-01-31:
yeah, I don't always keep the xci files up to date as new versions of vivado are released.  What you can do to update that in linux is run make fpga.xpr, then open the project in Vivado, upgrade the IP core, replace the xci file, and then run make.  Or you can manually add all of the files to vivado and upgrade the IP; that should also work.  The generated bit file should work at that point, assuming there are no network configuration issues on your host PC.  

If you're not seeing any characters echoed pack with netcat, then please check to make sure the network interface on your computer is configured correctly - it has to have some IP in the 192.168.1.0/24 range, and the host actually has to use that interface to talk to the board, so you may have to add an ARP entry or disconnect from your local LAN.  You can also try changing the IP address in the verilog to something else.  

Use wireshark to examine traffic to and from the board - you should see the ARP request and reply as well as the UDP data.  

If that doesn't help, the next thing to try is using the ILA to look at the interface to the MAC to make sure the traffic is making it all the way through to the board.

ranjanbl, 2020-01-31:
Hi Alex,
I set the host IP and it worked like a charm. Thank you so much.

closed 2020-01-31 by ranjanbl