alexforencich/verilog-ethernet #20: Recursive call in priority encoder doesn't seem to work in Synplify

chen-m76, 2020-02-13:
In priority_encoder.v, when WIDTH is greater than 2, I believe Synplify throws a warning and doesn't synthesize that section. In the ethernet stack, all of the WIDTHs (eth_arb_mux and ip_arb_mux) are equal to 2 so it didn't have any issues. However when using it standalone (axis_arb_mux), looks like that's causing some problems when WIDTH>2. Vivado seems to work fine. Anyone else seeing this issue?

alexforencich, 2020-02-13:
Interesting; I have heard Synplify totally blows every other synthesis tool out of the water.  Unfortunately, I don't have access to Synplify to test that issue myself.  Is there no setting or option that can be changed to enable synthesis of recursive module instances in Synplify?

chen-m76, 2020-02-13:
Could just be a Libero thing, opened the file imported into synplify
```
// Overrides src/third_party/eth_mac/priority_encoder.v
//
// Libero doesn't support recursive modules.

module priority_encoder #(
  parameter int WIDTH = 4,
  // LSB priority: "LOW", "HIGH"
  parameter LSB_PRIORITY = "LOW"
) (
  input  wire [WIDTH-1:0]         input_unencoded,
  output wire                     output_valid,
  output wire [$clog2(WIDTH)-1:0] output_encoded,
  output wire [WIDTH-1:0]         output_unencoded
);

generate
  if (WIDTH == 1) begin
    // one input
    assign output_valid = input_unencoded;
    assign output_encoded = 0;
  end else if (WIDTH == 2) begin
    // two inputs - just an OR gate
    assign output_valid = |input_unencoded;
    if (LSB_PRIORITY == "LOW") begin
      assign output_encoded = input_unencoded[1];
    end else begin
      assign output_encoded = ~input_unencoded[0];
    end
  end else begin
    initial $fatal(1, "[%m] Unsupported WIDTH=%d", WIDTH);
  end
endgenerate

// unencoded output
assign output_unencoded = 1 << output_encoded;

endmodule
```

I'll look around for an option (or maybe just run Synplify separately)

alexforencich, 2020-02-13:
Well, that explains quite a bit.  Issue is not Symplify, issue is actually Libero, coupled with a libero-specific "fix".  Maybe that file should be rewritten to not require recursion to make it portable to tools that don't correctly parse verilog.

chen-m76, 2020-02-13:
Actually, upon further investigation, someone else on the team made this fix and file. I think the original issue noted above still stands but Libero throws an error, it doesn't find and replace code in the file for you.

alexforencich, 2020-02-13:
Yeah, that's what I gather as well.  Didn't work on libero, someone made a band-aid "fix" instead of fixing it properly, then that "fix" broke things in synplify.  Point still stands that libero doesn't like the recursive priority encoder implementation, not sure what the best solution for that would be.  Probably just going to have to rewrite that module at some point to remove the recursion.

closed 2020-02-18 by chen-m76

alexforencich, 2020-08-18:
The priority encoder has now been rewritten to use loops instead of recursion.  Hopefully synplify will be happy with that.