alexforencich/verilog-ethernet #23: UDP invalid payload length

DeadMaX, 2020-03-01:
It seems that the UDP payload length is miss handeled.

If I have short frame, the udp_payload contains the padding.

o fix that, i change the flowing line into the file udp_ip_rx.v, from:

word_count_next = m_udp_length_reg - 16'd8;

to

word_count_next = m_udp_length_reg;

alexforencich, 2020-08-16:
With that change; the testbench does not pass?  The testbench checks for extra padding anyway.  And the UDP length field includes both the header (8 bytes) and payload, so 8 must be subtracted to get the payload length.  Please give me an ILA trace or test case that indicates what the issue is.

DeadMaX, 2022-02-18:
Hi Alex,

Sorry for the long delay, I completly forget to respond.

All is working fine with the last updates. Thank you for your support !

closed 2022-02-18 by DeadMaX