alexforencich/verilog-ethernet #30: VCU-118 10G and 25G Ethernet half expected data rate

MostParsingVex, 2020-04-10:
I may be doing something dreadfully wrong but I'm getting about half the expected data rates for the 10G and 25G QSFP Ethernet examples. (its okay, even half rate is more than enough for my application!)

The example projects appear to use all the QSFP pins on both connectors; is it expecting to flow data through both connectors? (I only have one connected; could this explain me getting half the expected data rate?)

alexforencich, 2020-04-10:
The example designs connect PHYs to all of the GTY channels, but only one of them is connected to the MAC and the rest of the design.  
How are you testing the data rate?

alexforencich, 2020-12-20:
I just found the issue; the register type was set incorrectly on the AXI stream switch that is used for inserting the gigabit port into the high speed path on the VCU108 and VCU118 designs.  The reg type that was selected was a non-skid-buffer, which uses one set of registers but has a throughput of half.  The correct reg type is a full skid buffer that provides full throughput.