alexforencich/verilog-ethernet #31: make build error for VCU118 projects

wrmarchetto, 2020-04-23:
When I try to run make on any of the VCU118 projects, I get the following error:
```
make[1]: Entering directory '<path>/verilog-ethernet/example/VCU118/fpga_25g/fpga'
make[1]: *** No rule to make target 'fpga.bit', needed by 'fpga'.  Stop.
```

I've added the following to my path:
C:\Xilinx\Vivado\2019.2\bin
C:\Xilinx\Vivado\2019.2\bin\unwrapped\win64.o
C:\Xilinx\Vivado\2019.2\scripts

Am I missing something in my path? Thanks!

alexforencich, 2020-04-23:
symlinks don't work in windows; you'll have to create a vivado project and add all of the files manually instead of using the makefile.

wrmarchetto, 2020-04-24:
Ah, didn't notice the symlinks. Git Bash supports symlinks if you run as administrator and execute `export MSYS=winsymlinks:nativestrict`. After doing that, make worked properly. Thanks!

closed 2020-04-24 by wrmarchetto