alexforencich/verilog-ethernet #32: Undriven PTP inputs when instancing 'eth_mac_1g'

egrigor, 2020-05-28:
First, thank you for sharing your valuable AXIS and Ethernet repositories.
I'm currently using your 'eth_mac_1g' MAC on a VCU118 platform, and noticed that you don't drive the 'tx_ptp_ts' and 'rx_ptp_ts' inputs when you instance this module in 'eth_max_1g_fifo' module.

I noticed this when simulating my design with your MAC and the 'gig_ethernet_pcs_pma'.  I'm currently forming my own raw Ethernet frames and sending these directly via the AXIS interface into 'eth_mac_1g' (bypassing all of the UDP/IP/ARP stack modules and also using Xilinx AXIS data FIFO instead of your CDC FIFOs) .  For some unknown reason, the PHY on the receive end of this link is reporting CRC errors.  I do see that you do compute the CRC and transmit to the 'gig_ethernet_pcs_pma' block via the 'gmii_txd' and 'gmii_tx_en' signals, but for some reason, when the FCS field is examined on the the receive side of the link (via Wireshark), it shows the FCS bits all set to 1's.  Not sure if this is related to the undriven PTP inputs (probably not), but I figured I'd let you know about this discrepancy.

Thank you

alexforencich, 2020-05-28:
PTP is disabled by parameter by default, and when PTP is disabled those inputs are ignored.  The PTP interface is not necessarily 100% complete, when it is then I will go add and tie off the unused connections to squelch a few warnings.  But the unconnected inputs will not affect the operation of the core.

egrigor, 2020-05-28:
Ok, thank you for the quick response.

closed 2020-05-28 by egrigor

alexforencich, 2020-05-28:
It would be good to get to the bottom of what is going on in your setup.  First, the PHY does not check the FCS, that's the job of the MAC.  You say that you are seeing an FCS of all 1s at the link partner?  What happens in simulation - do you see this all 1s FCS coming out of the MAC module?  What do you see if you add an ILA to monitor the TX GMII signals?  Also, one thing I should note about checking the FCS with wireshark: in general, the FCS field is checked and stripped by the NIC, in which case wireshark never sees it.  There can be a similar problem with checksum offloading, in which wireshark detects that the TCP/UDP checksums are wrong on all transmitted packets as it sees the packet before the NIC computes and updates the checksum.

egrigor, 2020-05-29:
Right.  In reality, I have been having lots of issues with the 'gig_ethernet_pcs_pma' to work with the TI PHY on the VCU118.  Now this is going beyond the scope of this original discussion.  Originally, I was using the Xilinx AXI 1G/2.5G Ethernet Subsystem, but the AXI-Lite FSM in Xilinx's example design failed to program the SGMII 625 MHz differential clock to stay on for much time after configuration.   The 625 MHz clock would not stay on.  I have a couple of colleagues who have also worked on this interface earlier.  One of them resorted to using an external 125 MHz clock to provide the needed reference clock to the Xilinx subsystem.  That approach worked up to point as we started to see frame dropouts starting at ~20 Mbps on this link that is supposed to be capable of going to 1 Gbps.  Looking over the Xilinx docs, I see that 125 MHz ref clocking is only available as an option for synchronous mode clocking which Xilinx clearly states is not supported on UltraScale+.   For Ultrascle+, they state that one must use asynchronous mode clocking, so I'm theorizing that this may be the cause of not being able to go much faster than ~20 Mbps with that approach.   So I decided to try your MAC and your mdio_master blocks, as another colleague was able to replicate a modified version of your VCU118 'fpga_1g' example design and test it successfully up to ~327 Mbps (we didn't try to go faster).  Now I'm trying to get this MAC to work on my design which has a different design flow and is SystemVerilog-based:  I use Synplify Premier for synthesis and Vivado 2019.1.3 for the backend implementation.  Getting the constraints to work from your XDC and TCL files especially for the FIFOs was not so straigthforward as Synplify's EDIF netlist does not match the cell names that you have in your TCL files.  I did make some modifications to those constraints with some limited success but instead opted to use the Xilinx XPM syncs and their axis asynchronous data FIFO for the CDC components.   Synplify Premier undestands the Xilinx XPM elements and also generates the necessary constraints for these.  Having successlully tackled the CDC concerns, I now see the FCS issue.   In simulation, I do see my frames pass the data fine to your MAC, out of the TX FIFO and see that your MAC appends the FCS to the Ethernet frames before transmitting them to the 'gig_ethernet_pcs_pma' phy_gmii_txd input.  I have not dropped an ILA on the interface between the MAC and the 'gig_ethernet_pcs_pma' yet as I already see my payload data arrive on the receiving end computer fine with the exception of the FCS field being set to all 1's.  Wireshark would not initially see these 'bad' frames at all, but I could tell that there were CRC errors at the NIC PHY/ASIC using the Linux command 'ifconfig p2p1' command.  Issuing the command 'ethtool -K p2p1 rx-fcs on rx-all on' tells the NIC to not discard the packets and to not discard the FCS field.  So, that's how I'm seeing the FCS bits set to all 1's.  It could still be that the Realtek NIC does not fully understand/support this feature, and is just sending filler 1's instead of the actual received FCS, or could be that Wireshark is doing what you stated above.  So, I'm still checking other ethtool option settings that might help in diagnosing this interface.

alexforencich, 2020-05-29:
Oh, that PHY chip is horribly annoying as it can't be pin-strapped into a sane SGMII configuration, unlike the Marvell PHYs that basically just work without having to jump through any hoops.  (On the flip side, you can actually get the datasheet for the TI part, as opposed to the Marvell parts where the datasheets are not public).  

Ouch, I did not realize that synplify would generate such different net names.  That does complicate things significantly.  I don't have access to synplify though, so I have no way of debugging that sort of issue.  Have you tried using my MDIO init code with the Xilinx MAC?  That might be worth a shot.  

From my experience with high-speed NICs, many 10G and faster NICs don't even provide the option for passing through the FCS or disabling checking the FCS.  I have not done much poking around with gigabit NICs.  It's entirely possible that something is not implemented correctly either in the NIC hardware or in the driver that's preventing that from working correctly.  Maybe try a different NIC and see if it behaves differently.

egrigor, 2020-05-29:
My sentiments exactly.  You get what you get with the TI PHY since it's on a Xilinx-designed board.  Yes, I had the same frustration with a Marvell Phy years ago where I could not even get a copy of their data sheet, so gave up on doing wired/copper-interface and switched to doing a SFP-based optical link.  Synplify's renaming is not that different than Vivado's, I think, since it just adds _0, _1, _2 to registered output nets when it replicates a register to reduce fanout. As for using your MDIO init code with Xilinx MAC, that thought actaully did cross my mind this morning, but not having any RTL visibilty into Xilinx's Ethernet subsystem block, I am not sure if there are any interdependcies between the clock used on their AXIL-based FSM and the main link's clock, but I will take a closer look.  Playing with another NIC & VCU118 pair is also an option as we have several of these lying around (but having to access all of these resources remotely in the current environment).  Thank you

egrigor, 2020-05-30:
@alexforencich  While chasing down the 32'hffff for the CRC, I dropped an ILA between your MAC and the 'gig_ethernet_pcs_pma' block to monitor the gmii_txd, gmii_tx_en and gmii_tx_er signals.  Surprisingly, the MAC IS driving the CRC value to all 1's at the end of the payload (top picture), which is the opposite of what I see in simulation (bottom picture).  So what the NIC PHY is reporting for the FCS at the other end of the link matches the ILA data.  I've also included portion of the code where I instance your MAC.  Any clues what may be causing this discrepancy?  Do I have something misconfigured for the MAC?  My AXIS frame size delivered to the MAC in this instance is 646 bytes, if that makes any difference.  

[image]

[image]

eth_mac_1g #(
            .ENABLE_PADDING     (1),
            .MIN_FRAME_LENGTH   (64)
        )
        eth_mac_1g_inst (
            .tx_clk             (phy_gmii_clk_int),
            .tx_rst             (rst_int),
            .rx_clk             (phy_gmii_clk_int),
            .rx_rst             (rst_int),
            /*
             * AXI input
             */
            .tx_axis_tdata      (tx_axis_mac.tdata), 
            .tx_axis_tvalid     (tx_axis_mac.tvalid),
            .tx_axis_tready     (tx_axis_mac.tready),
            .tx_axis_tlast      (tx_axis_mac.tlast), 
            .tx_axis_tuser      (tx_axis_mac.tuser),
            /*
             * AXI output
             */
            .rx_axis_tdata      (rx_axis_mac.tdata), 
            .rx_axis_tvalid     (rx_axis_mac.tvalid),
            .rx_axis_tlast      (rx_axis_mac.tlast), 
            .rx_axis_tuser      (rx_axis_mac.tuser), 
            /*
             * GMII interface
             */
            .gmii_rxd           (phy_gmii_rxd_int),  
            .gmii_rx_dv         (phy_gmii_rx_dv_int),
            .gmii_rx_er         (phy_gmii_rx_er_int),
            .gmii_txd           (phy_gmii_txd_int),  
            .gmii_tx_en         (phy_gmii_tx_en_int),
            .gmii_tx_er         (phy_gmii_tx_er_int),
            /*
             * PTP
             */
            .tx_ptp_ts            ('0),                         // input  wire [TX_PTP_TS_WIDTH-1:0]                                                                                                               
            .rx_ptp_ts            ('0),                         // input  wire [RX_PTP_TS_WIDTH-1:0]   
            .tx_axis_ptp_ts       (),                           // output wire [TX_PTP_TS_WIDTH-1:0]   
            .tx_axis_ptp_ts_tag   (),                           // output wire [TX_PTP_TAG_WIDTH-1:0]  
            .tx_axis_ptp_ts_valid (),                           // output wire                         
            /*
             * Control
             */
            .rx_clk_enable      (phy_gmii_clk_en_int),
            .tx_clk_enable      (phy_gmii_clk_en_int),
            .rx_mii_select      (1'b0),
            .tx_mii_select      (1'b0),
            /*
             * Status
             */
            .tx_error_underflow (),
            .rx_error_bad_frame (),
            .rx_error_bad_fcs   (),
            /*
             * Configuration
             */
            .ifg_delay          (8'd12)
        );

alexforencich, 2020-05-30:
Ah.  What could be going on is that synplify does not like something about my unrolled LFSR module and as a result that module is outputting all zeros, which gets inverted to all ones.  In the middle of lfsr.v there is a commented out loop with a $display in it.  Uncomment that loop and let me know what synplify prints during synthesis.

egrigor, 2020-05-30:
Here is the portion pertaining to the lfsr from Synplify:

alexforencich, 2020-05-30:
Uncomment this block and run it again: https://github.com/alexforencich/verilog-ethernet/blob/master/rtl/lfsr.v#L352

egrigor, 2020-05-30:
Yes, that's exactly what I did, and then got the compile log that I posted above (only lfsr-related logs).  Note that I edited that original post as I found some more logs related to the lfsr (separated by ...)

egrigor, 2020-05-30:
Actually, looks like Synplify optimizes the lfsr module out (last note in the compile log above) since it does not drive other instances

alexforencich, 2020-05-30:
Well, I am expecting to see a big block of numbers from that loop.  It should print this:

    00000000000000000000000100000100 00000100
    00000000000000000000001000001001 00001001
    00000000000000000000010000010011 00010011
    00000000000000000000100000100110 00100110
    00000000000000000001000001001101 01001101
    00000000000000000010000010011010 10011010
    00000000000000000100000000110000 00110000
    00000000000000001000000001100001 01100001
    00000000000000010000000011000010 11000010
    00000000000000100000000010000000 10000000
    00000000000001000000000000000100 00000100
    00000000000010000000000000001000 00001000
    00000000000100000000000000010001 00010001
    00000000001000000000000000100011 00100011
    00000000010000000000000001000110 01000110
    00000000100000000000000010001100 10001100
    00000001000000000000000000011101 00011101
    00000010000000000000000000111011 00111011
    00000100000000000000000001110111 01110111
    00001000000000000000000011101110 11101110
    00010000000000000000000011011000 11011000
    00100000000000000000000010110100 10110100
    01000000000000000000000001101100 01101100
    10000000000000000000000011011000 11011000
    00000000000000000000000010110101 10110101
    00000000000000000000000001101111 01101111
    00000000000000000000000011011111 11011111
    00000000000000000000000010111010 10111010
    00000000000000000000000001110001 01110001
    00000000000000000000000011100011 11100011
    00000000000000000000000011000011 11000011
    00000000000000000000000010000010 10000010
    00000000000000000000000100000100 00000100
    00000000000000000000001000001001 00001001
    00000000000000000000010000010011 00010011
    00000000000000000000100000100110 00100110
    00000000000000000001000001001101 01001101
    00000000000000000010000010011010 10011010
    00000000000000000100000000110000 00110000
    00000000000000001000000001100001 01100001
    00000000000000010000000011000010 11000010
    00000000000000100000000010000000 10000000
    00000000000001000000000000000100 00000100
    00000000000010000000000000001000 00001000
    00000000000100000000000000010001 00010001
    00000000001000000000000000100011 00100011
    00000000010000000000000001000110 01000110
    00000000100000000000000010001100 10001100
    00000001000000000000000000011101 00011101
    00000010000000000000000000111011 00111011
    00000100000000000000000001110111 01110111
    00001000000000000000000011101110 11101110
    00010000000000000000000011011000 11011000
    00100000000000000000000010110100 10110100
    01000000000000000000000001101100 01101100
    10000000000000000000000011011000 11011000
    00000000000000000000000010110101 10110101
    00000000000000000000000001101111 01101111
    00000000000000000000000011011111 11011111
    00000000000000000000000010111010 10111010
    00000000000000000000000001110001 01110001
    00000000000000000000000011100011 11100011
    00000000000000000000000011000011 11000011
    00000000000000000000000010000010 10000010

If that doesn't appear or if it's all zeros, then something is very wrong.  The message "Within an initial block, only Verilog force statements and memory $readmemh/$readmemb initialization statements are recognized, and all other content is ignored." leads me to believe that synplify may be ignoring all of that code, which would explain the issue.

alexforencich, 2020-05-30:
Another thing to try is to swap out the lfsr instance for the old generated code: https://github.com/alexforencich/verilog-ethernet/blob/47ca9a8725dba00990341470b6af37148c50c5ec/rtl/eth_crc_8.v

egrigor, 2020-05-30:
Yeah, I don't see anything like that anywhere in the compile log.  I don't' think Synplify would print anything with $display in it as it is not a supported function and is ignored.

egrigor, 2020-05-30:
Ok, I will try that and let you know.  Thanks

alexforencich, 2020-05-30:
Wtf, display is not a supported function?  Vivado and Quartus are both perfectly fine with it, it can be a lifesaver for debugging constant elaboration issues.  But it seems synplify may be even worse than quartus when it comes to constant elaboration.  I may just have to add a note to the readme that synthesis with synplify is not supported.

egrigor, 2020-05-30:
I think I now follow what you were stating earlier and see what's going on with the synthesis of the lfsr module.  The contents of the initial block are all ignored (lines 211-355), so the 'lfsr_mask_state' and 'lfsr_mask_data' registers that are used in the synthesizable section of the code don't ever change from being zeros.  With those being at zero, there's not much left to synthesize in the LOOP portion, so effectively the whole thing get's optimized out.  It's not the $display that's the issue, but rather the synthesizable portion of the code is depending on values that come from the initial block section which are being completely ignored.
I just finished synthesizing and implementing with the 'eth_crc_8' instead of the 'lfsr' module.  Now, the FCS issue appears to be fixed as I can see the CRC on the ILA as well as on the receiving end of the link via Wireshark.  Neither ifconfig nor wireshark are complaining about the FCS any longer.  Thank you for your help in resolving this issue.

alexforencich, 2020-05-30:
If the initial block computes constants, then it is synthesizable via constant elaboration.  However, it seems synplify does not support constant elaboration, which means it is impossible to parametrize the computation of initial values.  That is a major shortcoming of synplify, and I should add a disclaimer that synthesis with synplify is not supported for any of my code until they correct that issue.

egrigor, 2020-05-30:
You can use parameter and localparams to compute initial values.  You can even have functions and nested functions in the parameter section that compute new parameter values that can then be used in the rest of the code.  I've done this before where I had to do a fairly extensive set of parameter initializations to compute to use further down in my code.  You just need to be aware of the order of parameter declarations when you define each one.  I don't think Synplify is unique here in this regard and is just doing what most major EDA, non-FPGA vendor-provided synthesis tools do, not to mention Synplify and Design Compiler, which would behave the same way, are from Synopsys who is a long-leader in the logic synthesis area.

alexforencich, 2020-05-31:
OK.  In that case, how do I return four 2D arrays from the same function in verilog-2001 so that ISE, Quartus, Vivado, and Synplify are all happy with the result?

alexforencich, 2020-05-31:
After taking another look at the module, it looks like if I concatenate the inputs and the outputs, I only need one 2D array, which I can output as a single 1D wire of (LFSR_WIDTH+DATA_WIDTH)*(LFSR_WIDTH+DATA_WIDTH).  A bit excessive, but if all of the tools are happy with that, then I guess it might be possible to convert lfsr.v to use parameters.

egrigor, 2020-05-31:
Sounds like a good plan

egrigor, 2020-09-18:
Hi Alex,
I'm now using your 'udp_complete' stack to incorporate UDP/IP/ARP into the 1Gig Ethernet implementation.  As mentioned above, I'm using Synplify for synthesis and as such Synplify ignores all code, except for memory initialization, specified in any Verilog initial block.
Previously you provided me with etc_crc_8.v as a workaround for the lfsr.v module as the lfsr module has an initial block that Synplify just ignores.  The replacement etc_crc_8.v module works fine for the case of Ethernet frames, however I noticed that the arp_cache.v file instanced in 'udp_complete' is also using the lfsr.v but for 32-bit CRC calculation.  Do you have an equivalent eth_crc_32.v file that I can maybe use as a substitiute for the two lfsr instances (rd_hash and wr_hash) inside the arp_cache.v file?  So far that's the only place in udp_complete.v where I noticed you use the lfsr.v.   I'm having intermittent startup issues with UDP implementation with the target computer, and I believe this issue is the reason for it.  Thank you in advance.  Edwin

alexforencich, 2020-09-18:
I do not, but you may be able to generate one with this: https://github.com/alexforencich/fpga-utils/blob/master/crcgen.py .  This is the script used to generate the original eth_crc_*.py modules.

egrigor, 2020-09-18:
Actually, I just checked the same directory where you store eth_crc_8.v, and I see an eth_crc_32.v file there.  You don't think that one will work?

alexforencich, 2020-09-18:
Actually, if the interface is the same, then that file might also work.

egrigor, 2020-09-18:
I will check.  Thank you

egrigor, 2020-09-18:
I checked and the interfaces are the same.  I rebuilt the FPGA code using eth_crc_32.v instead of lfsr.v in arp_cache.v, and UDP appears to be working reliabily now.  Thank you

alexforencich, 2020-09-19:
Good to hear!