alexforencich/verilog-ethernet #34: Vivado block diagram, recursive priority encoder

Neywiny, 2020-06-24:
I successfully built the example project for my board with Vivado, but now I need to integrate it with the rest of my system (Microblaze). Just like [this issue here](https://github.com/alexforencich/verilog-ethernet/issues/20) Vivado is complaining about the recursive definition of the priority encoder when trying to import the modules into the block diagram. From what I understand about [this question](https://github.com/alexforencich/verilog-ethernet/issues/10) I only need the eth_mac_1g_rgmii_fifo and the udp_complete (and the TCL files, I just copied all the ones used in the example project). So potentially I could just rewrite the module myself? I know back in Feb there was talk of doing so. Doesn't seem like it would be too terribly difficult, but I doubt my abilities when it comes to high speed stuff.

alexforencich, 2020-08-18:
The priority encoder has now been rewritten to use loops instead of recursion.  However, I don't know if this will completely satisfy Vivado's IPI flow.

Neywiny, 2020-08-18:
So the priority encoder is now allowed to be added to the block diagram, but there are still some things it doesn't like. It can't find the Altera stuff so I removed that from the i/oddr, but overall it seems to be having issues with the structure of the project. Lots of ambiguous reference errors when trying to add things. Seems like this specific issue is fixed, but my goal was not accomplished.

alexforencich, 2020-08-18:
It seems like the IPI flow basically tries to import everything recursively, which is problematic as you can't simply write a wrapper at the top-level and import that; the whole thing has to be packaged as an IP core which involves jumping through a large number of hoops.  I think the conclusion here is that the IPI flow is terrible, just like most other graphical design tools.

Neywiny, 2020-08-18:
I think sometimes it's helpful. I haven't used it extensively before this project but after getting accustomed to it it does make a lot more sense. There's just too many things where once you deviate from the GUI designer's intentions nothing works anymore. Somehow it needs even more warnings.

alexforencich, 2020-08-18:
Right, if you use it exactly as intended, it sort of works.  But if you try to do something strange, then you run in to issues.  Now, if the vivado devs made it more tolerant of this sort of thing, it might be a different story.  But I'm not going to rewrite half of the codebase and purge Intel device support just to appease IPI.

Neywiny, 2020-08-18:
I completely agree. I only asked for the priority encoder because it seemed like other flows needed it to be rewritten. The rest of this stuff I suppose would be nice to have as an option to replace the Xilinx Ethernet IP, but it seems like a lot of work would need to be done to get that working.

alexforencich, 2020-08-18:
Yeah, reworking the priority encoder was definitely a reasonable request, and I also figured maybe rewriting that would enable things to work better with on zynq devices as xilinx is really pushing the IPI flow on those devices.  However, even with the priority encoder rewritten, I am running in to a multitude of other issues with the IPI flow.  The really infuriating part is that the error messages you get from IPI are worthless.  They don't even tell you what the problem modules are, what the conflicts are, etc.  And it marks modules as incompatible without any explanation as to what it doesn't like about them.  And it doesn't like many of my parametrization schemes.  

Perhaps what I could look in to doing is packaging some of the blocks as IP cores that the IPI flow will be more happy with.  Makes a reasonable amount of sense for things like the Ethernet MACs.  But this isn't a high priority for me as I avoid the IPI flow as much as possible, especially now that I have figured out that it takes about 5 lines of TCL to wrap an IPI-only block with a verilog wrapper so it can be directly instantiated in a verilog module without any BS.

Neywiny, 2020-08-18:
I think I have to use the IPI flow for my project at the moment so if I figure it out before my deadline I'll maybe try some git magic to pull request or branch or something. The error messages are very annoyingly absent. I can't remember where I saw the one about the recursive definition of the priority encoder, but it took a lot of digging. Honestly may have just found it on some random forum about another project. I have had some success with running the TCL command to just add the module and see what it says, but not much. One of those things where something is throwing an error but it's not getting propagated right so if the middle man can be cut out there might be more info.

I think once this game of whack-a-mole is over your IP could be very useful to a lot of projects, it seems like a lot of people want to use an open source Ethernet IP but surprise surprise that's not an easy ask.