alexforencich/verilog-ethernet #35: PC Ethernet connection settings

bashidagha, 2020-07-18:
I used KC705 Ethernet code to implement Ethernet on my custom Artix board. I remove button and uart interface of code, because i haven't them on my board.
I used ila to see output of `phy_rxd` when running on the board.
After programming and running, I set Ethernet cable with below code: (eth0 is board connection)
`ifconfig eth0 192.168.1.128 netmask 255.255.255.0`
and set configuration of ethernet connection to `Automatic(DHCP)`
but when i echo something through below terminal command:
`netcat -u 192.168.1.128 1234`
nothing coming on `phy_rxd` .
Is something wrong with my setting?

alexforencich, 2020-07-18:
Well, first thing you need to do is set an IP address other than 192.168.1.128 on that interface.  I usually use 192.168.1.199; doesn't really matter so long as it's in the same subnet as 192.168.1.128/24 but is a distinct address.  Also, you may run in to issues if you're connected to a 192.168.1.0/24 network on a different interface - in which case, you might have to temporarily disconnect from that network or do something else to isolate the interfaces.  If that doesn't help at all, then I'm going to need more information about your setup.

bashidagha, 2020-07-18:
Thanks @alexforencich  for your fast response, I do above but the problem stills.
I attached my modified code. The difference from your code:

1- use clocking wizard to generate an 200 M and 125 M clock form my onboard 40 M clock.
2- remove buttons and uarts

I ran above code on my vivado and monitor `phy_rxd`. I set my eth0 connection to Automatic(DHCP) and ran something like below to assign static ip:
`sudo ifconfig enp7s0 192.168.1.199 net mask 255.255.255.0`

Then use below command to echo to fpga:(I'm not sure about correct port and ip)
`netcat -u 192.168.1.128 1234`

 Also my fpga directly connected to my PC via Ethernet cable.

bashidagha, 2020-07-18:
Also i should say that my connection is not available even after fpga programmed.
[image]

alexforencich, 2020-07-18:
What do you see if you run tcpdump -i enp7s0 while attempting to talk to the board?  Does tcpdump capture any traffic?

bashidagha, 2020-07-19:
nothing come unfortunately!
```
sudo  tcpdump -i enp7s0
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on enp7s0, link-type EN10MB (Ethernet), capture size 262144 bytes
```

alexforencich, 2020-07-19:
Well, if nothing is being sent to the board, then the board certainly isn't going to receive anything.  Try this script to isolate the interface to prevent any other linux tools from screwing with it:

    #!/bin/bash

    ns=test
    dev=$1

    if [ -z "$dev" ]; then
        echo "Error: no device specified"
        exit 1
    fi

    ip netns add $ns
    ip link set dev $dev netns $ns
    ip netns exec $ns ip link set dev $dev up
    ip netns exec $ns bash
    ip netns del $ns

Save this as dev-netns-shell.sh, then run it as sudo ./dev-netns-shell.sh enp7s0.  This will add a network namespace, place enp7s0 in this namespace, and then drop you into a shell running in that namespace.  From that shell, set up the interface IP and try running netcat and see if the board gets any traffic.

bashidagha, 2020-07-19:
Thanks @alexforencich , But I can't get any traffic.
It's so weird for me, i can not see traffic with wire shark also.
I used MDIO of PHY for configuration with hdl codes and sure that PHY is OK.
My PHY is 88e1111 similar to KC705 board.
It's become more weird, because when i echo data through below command:
` netcat -u 192.168.1.128 1234`
i can't see any thing on gmii_rx pins of 88e1111 with ila.
Do you have any suggestion for me?

alexforencich, 2020-07-19:
Not sure at this point.  If the link is up at the PHY layer on both ends, but linux isn't sending data, I don't know what the problem is.  Maybe I could figure it out if I was sitting down in front of it, but I'm not so I'm limited to providing general advise.  My best guess now is that maybe the link is not actually up at the PHY layer and hence linux isn't sending anything.  One of your ifconfig outputs is showing TX errors, which should not happen under normal circumstances, I'm not sure if that might be related or not.

alexforencich, 2020-07-19:
Well, one last thing you can try is raw sockets: https://stackoverflow.com/questions/1117958/how-do-i-use-raw-socket-in-python.  Ignore the ethtool and checksum stuff as you want the NIC to generate and insert the FCS, but this is one way to send an arbitrary packet out of a specific interface.  If that doesn't work, then there is something misconfigured that you haven't told me about or something seriously wrong with the linux installation or the network hardware.

bashidagha, 2020-07-21:
@alexforencich  the 88e1111 PHY has a packet generator feature. When i send packet with that i can get them in wireshark.
I tried to separate the part's can send raw data through ethernet (Not UDP) and capture them on wireshark. but it's a few complex code, did you have this part seprately?

alexforencich, 2020-07-21:
Look, I can help you debug an existing example design on standard hardware, or issues that are internal to the stack and can be replicated in simulation or on standard hardware.  If you're running in to board-level issues on a custom board, then we're going to have to set up some sort of a consulting agreement, and you're going to have to provide hardware.  Otherwise I'm just shooting in the dark.

closed 2020-07-22 by bashidagha