alexforencich/verilog-ethernet #36: Which NIC did you use for the AU50 example?

TaekyungHeo, 2020-07-23:
AU50 example does not work in my setup.

I think that the reason is my NIC (Intel XL710 for 40GbE QSFP+).
Which NIC did you use for the AU50 example?
Do you have any suggestions for trouble shooting?

alexforencich, 2020-07-23:
I used a Mellanox ConnectX-5.  It's possible that the Mellanox NICs are more conducive to figuring out the correct settings when the link partner doesn't support AN.  Does ip link/ifconfig report that the link is up?  What does ethtool report for the XL710?  Have you tried using ethtool to force the XL710 to run at 10G instead of 40G?  Do you have a QSFP+ to SFP+ breakout cable and 10G SFP+ NIC so you can test with a 10G NIC?

TaekyungHeo, 2020-07-23:
Thanks a lot, Alex!.

I tried to use `ethtool` to change the settings of my NIC, but it didn't work well.
I am not sure whether the NIC supports changing those settings (AN, speed).
It would be a good option to purchase a Mellanox ConnectX-5.

closed 2020-07-23 by TaekyungHeo

TaekyungHeo, 2020-07-23:
I could find a ConnectX-4 in my lab, and it works well.
I had to change the link type of NIC from InfiniBand to Ethernet.

```
$ ethtool enp94s0f0
Cannot get wake-on-lan settings: Operation not permitted
Settings for enp94s0f0:
        Supported ports: [ Backplane ]
        Supported link modes:   1000baseKX/Full
                                10000baseKR/Full
                                40000baseKR4/Full
                                40000baseCR4/Full
                                40000baseSR4/Full
                                40000baseLR4/Full
                                56000baseKR4/Full
                                25000baseCR/Full
                                25000baseKR/Full
                                25000baseSR/Full
        Supported pause frame use: Symmetric
        Supports auto-negotiation: Yes
        Supported FEC modes: None BaseR
        Advertised link modes:  1000baseKX/Full
                                10000baseKR/Full
                                40000baseKR4/Full
                                40000baseCR4/Full
                                40000baseSR4/Full
                                40000baseLR4/Full
                                56000baseKR4/Full
                                25000baseCR/Full
                                25000baseKR/Full
                                25000baseSR/Full
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: Yes
        Advertised FEC modes: None
        Speed: 10000Mb/s
        Duplex: Full
        Port: Direct Attach Copper
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        Current message level: 0x00000004 (4)
                               link
        Link detected: yes

$ sudo ethtool -s enp94s0f0 autoneg off speed 10000 duplex full
Cannot get wake-on-lan settings: Operation not permitted
Settings for enp94s0f0:
        Supported ports: [ Backplane ]
        Supported link modes:   1000baseKX/Full
                                10000baseKR/Full
                                40000baseKR4/Full
                                40000baseCR4/Full
                                40000baseSR4/Full
                                40000baseLR4/Full
                                56000baseKR4/Full
                                25000baseCR/Full
                                25000baseKR/Full
                                25000baseSR/Full
        Supported pause frame use: Symmetric
        Supports auto-negotiation: Yes
        Supported FEC modes: None BaseR
        Advertised link modes:  10000baseKR/Full
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: No
        Advertised FEC modes: None
        Speed: 10000Mb/s
        Duplex: Full
        Port: Direct Attach Copper
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: off
        Current message level: 0x00000004 (4)
                               link
        Link detected: yes

$ sudo ifconfig enp94s0f0 192.168.1.129
$ nc -u 192.168.1.128 1234
hi
hi
```

alexforencich, 2020-07-23:
Nice!  Yeah, I also had to flip a few connectx-5 NICs from IB to Eth recently; that's a bit of a pain but at least it's an option!  I also have a few connectx-4 NICs, I'm not sure if I tested the U50 design with a connectx-4 or a connectx-5.  I have had reasonably good luck with not having to manually force a specific configuration with the mellanox NICs, they seem to be pretty good about figuring things out.  Well, they do cause trouble with 10G designs on ultrascale as the NIC tries 25 Gbps first and gets the equalizers on the FPGA all screwed up so the RX side of the link won't work without a reset.  Fixing that is on my to-do list.  Seems to be less of an issue on ultrascale+.  At any rate, it's good to hear you got it working!