alexforencich/verilog-ethernet #37: Problem with sythesizing - VCU108 - fpga_1g

seva-r, 2020-08-12:
Hi,

I am trying to implement the 1g version on the VCU108 evaluation board. I manually added all the files to the project as I am using Windows machine.

I added the files as per the make file which has links to all the files needed for fpga-1g version. ( Folder ..../fpga-1g/fpga)

When i try to synthesize i get the error saying that the get_property should have atleast one object and is referring to both the tcl scripts eth_mac_fifo and axis_async_fifo.

It looks the hardware is not being generated only. I am not sure about the problem. I am I missing anything. I am using Vivado version 2019.2

When I disable both the tcl scripts I am sucessfully able to generate the bit file but it doesn't work. The IP 192.168.1.128 is reachable but nothing is being echoed back.

Can you please help me understand what can be the issue.

Thanks,

 Ramu

alexforencich, 2020-08-13:
Usually when I see that error, something is either screwed up with clocking somewhere and/or a large portion of the design was optimized away.  Did you modify the verilog at all, or did you import all of the files as-is?  Also, you may need to run the IP TCL files via the TCL console or something in vivado in order to create the IP cores.

seva-r, 2020-08-13:
Thank you for the reply. I found the issue. My constraints file were not in the correct order, hence the error. I corrected it and the design simulated fine. But still it doesn't work in the hardware. The IP is reachable when I check it. But for some reason I cannot loopback.

I am trying the loopback in a Linux machine. Regarding the IP cores I see only one TCL file that generates the PCS/PMA core. Are there other TCL files too..??

seva-r, 2020-08-13:
hi it looks all this time i was not matching the gateway ip with my host PC ip.. It works fine now..

closed 2020-08-13 by seva-r