alexforencich/verilog-ethernet #39: ENABLE_PADDING(0) leads to failure

qrp73, 2020-08-15:
I tested example based on ATLYS/fpga code. It works, but it add zero padding to the end of udp packet. I don't need it. 

In order to disable unwanted padding, I set ENABLE_PADDING(0) for eth_mac_1g_gmii_fifo. Unfortunately it leads to complete fail. Loopback example don't works with ENABLE_PADDING(0) and stops to receive payload from incoming udp packets (first byte of the udp payload is not passed to the led display).

Please fix it.

alexforencich, 2020-08-15:
Most likely with padding disable, the ARP response is being dropped due to being a runt frame shorter than 64 bytes.

qrp73, 2020-08-16:
> Most likely with padding disabled, the ARP response from the board is getting dropped somewhere due to being a runt frame shorter than 64 bytes. And it's not being dropped by the FPGA in this case

Well, to be more clear, I'm using example code with loopback on port 1234 from verilog-ethernet/example/ATLYS folder. I adapted the code for my starter-kit board QMTECH CYCLONE_IV running on EP4CE15F23 and RTL8211EG GMII. The changes is related to fpga.v only, where I implemented setup of 125 MHz pll and proper pin assignments.

By default example works well, it responds on ARP requests and sends back UDP packet from port 1234 with no issue.

But when I set ENABLE_PADDING(0) for eth_mac_1g_gmii_fifo in fpga_core.v, it stops to work.

I investigated the issue and found that there are two cases:
1) FPGA is connected to PC through router. In such case PC don't see any respond from FPGA on ARP request from PC. The PC sends ARP requests and there is no response.

2) FPGA is connected to PC directly. In such case the PC can see ARP response from FPGA, but don't see UDP response. FPGA can see UDP request, it shows first payload byte on led indicator. But there is no UDP response in wireshark.

So, it looks that the problem with transmit packet code on FPGA side when ENABLE_PADDING = 0. 

I should note, that my PC responds to ARP requests with no padding and router can accept ARP packets with disabled padding with no issue. But for some unknown reason it drops ARP response from FPGA board and don't route it to PC. I can see ARP response from FPGA when FPGA is connected directly to PC in Wireshark. At a glance ARP response looks ok, but I didn't checked crc.

This is strange why my router can accept ARP response with no padding from my PC and don't want to accept the same ARP response with no padding from FPGA. Is it possible that FPGA sent GMII packet in such way so the router thinks that it is malformed?

Probably for UDP response there is the same issue with packet transmit. It looks that FPGA transmit the UDP response, but since it is malformed, it is just dropped by PC network adapter and it even don't appears on Wireshark capture.

alexforencich, 2020-08-16:
Presumably you're using wireshark, tcpdump, or some similar software application to capture the packet data on your PC, correct?  In this case, you're seeing TX packets before they arrive at the NIC driver.  Either the NIC driver or the NIC hardware will pad the frames out to 64 bytes before they hit the wire.  So the outgoing UDP and ARP packets that you see that don't appear to have any padding will actually have that padding inserted before they leave the NIC.  

That's correct, a packet of less than 64 bytes is considered an error and is generally dropped by the NIC or by the OS at some level in the networking stack.

qrp73, 2020-08-16:
yes, I don't see packets with no padding coming from router. So, I think you're right, outgoing packets from PC which don't appear to have any padding will actually have that padding inserted before they leave the NIC.

if we take this into account, all works as expected, thanks for help :)

closed 2020-08-16 by qrp73