alexforencich/verilog-ethernet #40: Is there any example for UDP stack + DHCP?

qrp73, 2020-08-16:
udp_complete.v works ok, but it would be great to have DHCP support in order to obtain IP address automatically from the router. Is there any example? Thanks

alexforencich, 2020-08-16:
I don't have a DHCP client implementation at the moment.  Probably would not be all that complicated to implement as DHCP pretty much just uses UDP, the trick is to get the resource utilization low as it's only used very occasionally for configuration and as such does not require high performance.  This might be a good application for a small soft processor core or similar.

ffeldbauer, 2022-11-16:
Sry to revive this old issue, but I'm also interested in a DHCP client on top of your core.
Using the core with a static IP works just fine in my tests, but as I have to use the FGPA in different networks which does not work with static IP address.
Did anyone try to implement a DHCP client on top of this project already?

The DHCP Discover and Request packages have to be send to the IP `255.255.255.255` with destination MAC `ff:ff:ff:ff:ff:ff`.
Is this supported by your ARP module?