alexforencich/verilog-ethernet #41: AU50 and AU280 build error

kimanha, 2020-09-04:
Hello,

I am trying to build AU50& AU280 and found build error.

INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/opt/Xilinx/Vivado/2019.1/data/ip'.
WARNING: [IP_Flow 19-3664] IP 'gtwizard_ultrascale_0' generated file not found '/home/kha/fpga/verilog-ethernet/example/AU50/fpga_10g/fpga/fpga.srcs/sources_1/ip/gtwizard_ultrascale_0/gtwizard_ultrascale_0.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'gtwizard_ultrascale_0' generated file not found '/home/kha/fpga/verilog-ethernet/example/AU50/fpga_10g/fpga/fpga.srcs/sources_1/ip/gtwizard_ultrascale_0/gtwizard_ultrascale_0_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'gtwizard_ultrascale_0' generated file not found '/home/kha/fpga/verilog-ethernet/example/AU50/fpga_10g/fpga/fpga.srcs/sources_1/ip/gtwizard_ultrascale_0/gtwizard_ultrascale_0_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'gtwizard_ultrascale_0' generated file not found '/home/kha/fpga/verilog-ethernet/example/AU50/fpga_10g/fpga/fpga.srcs/sources_1/ip/gtwizard_ultrascale_0/gtwizard_ultrascale_0_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'gtwizard_ultrascale_0' generated file not found '/home/kha/fpga/verilog-ethernet/example/AU50/fpga_10g/fpga/fpga.srcs/sources_1/ip/gtwizard_ultrascale_0/gtwizard_ultrascale_0_sim_netlist.vhdl'. Please regenerate to continue.
# reset_run synth_1
# launch_runs synth_1
ERROR: [Common 17-70] Application Exception: Failed to launch run 'synth_1' due to failures in the following run(s):
gtwizard_ultrascale_0_synth_1
These failed run(s) need to be reset prior to launching 'synth_1' again.

INFO: [Common 17-206] Exiting Vivado at Fri Sep  4 19:46:42 2020...
../common/vivado.mk:95: recipe for target 'fpga.runs/synth_1/%.dcp' failed
make[1]: *** [fpga.runs/synth_1/%.dcp] Error 1
make[1]: Leaving directory '/home/kha/fpga/verilog-ethernet/example/AU50/fpga_10g/fpga'
Makefile:14: recipe for target 'fpga' failed
make: *** [fpga] Error 2

alexforencich, 2020-09-05:
Looks like you did not include the complete output, presumably there were some messages printed earlier that could indicate why generation failed.  What happens if you open the project in vivado and attempt to regenerate the IP manually?

However, I'm not sure if the U50 and U280 are supported in Vivado 2019.1; I think you need at least 2019.2 or 2020.1 for those boards.  The problem could simply be missing device support.

kimanha, 2020-09-05:
It working with Vivado 2020.1, and both projects successfully generated bit files.
I have the Alvio280 card and do you have a test procedure or any reference for test?

Thanks,

closed 2020-09-05 by kimanha

alexforencich, 2020-09-05:
I usually just use netcat in UDP mode as a quick sanity check for the example designs.