alexforencich/verilog-ethernet #42: Arty test, can't see loopback message

kimanha, 2020-09-06:
Hello,

I am trying to test the ARTY board loopback test.
Firstly, ping is not supported?

I use ‘netcat’ command you recommended at README.md.

After connection, I typed 1,2,3…8 number then I can see the LD2,3 ON and LD4,5,6,7 changing the same input number
But only 10 or 11 times working, then it has frozen and needs to hard reset.

Also, I typed string message, but I can’t see the echo string.

Please advise what I missing?

Thanks,

alexforencich, 2020-09-06:
ICMP (and hence ping) is not currently supported.  

Sounds like ARP isn't working correctly - the FPGA attempts to carry out an ARP request against your computer, and this fails for some reason, and the TX path on the FPGA is blocked while the FPGA attempts the ARP queries.  This could be caused by a host networking configuration issue.  How are you connecting the FPGA to your PC, and is your PC connected to any other networks?  If you have the board on a dedicated NIC and you're connected to wifi and have an address in the 192.168.1.x range, this can cause routing issues and you may need to disconnect from wifi to talk to the board.  Alternatively, you can use network namespaces to isolate the interface.

kimanha, 2020-09-07:
Hello,

I tried to connect the PC to ARTY directly and used an Ethernet switch with a disabled Wifi connection.
With an Ethernet switch, I connected ZCU102 for other TCP/IP communications it working fine.

I tested Ubuntu for netcat command and Widows UDP echo client program, both are the same result.
I can’t see the receiving data from the ARTY board.
I will try the Nexys Video board sometime next week.

Thanks,

alexforencich, 2020-09-07:
I just tested the current design on my Arty board, and it works fine.  I am going to go ahead and polish my netns script and get that in the repo as that is a very useful tool for testing and debugging the design.

kimanha, 2020-09-07:
Hello,

Thank you for the update.

It was my fault for the network configuration issue.
My network setup is used 10.0.142.xxx for my other system development environment.

Before the gateway is configured 192.168.1.1
because my other system used this gateway.

Now I changed gateway 10.0.142.1
wire [31:0] local_ip    = {8'd10, 8'd0, 8'd142,   8'd128};
wire [31:0] gateway_ip  = {8'd10, 8'd0, 8'd142,   8'd1};
wire [31:0] subnet_mask = {8'd255, 8'd255, 8'd255, 8'd0};

[cid:image001.png@01D68543.B633A7F0]

Thank you for your support.

alexforencich, 2020-09-07:
Good to hear you got it working!  I just added my netns shell script, which is useful for testing the verilog-ethernet design when there are possible networking conflicts.  See

https://github.com/alexforencich/verilog-ethernet/blob/master/scripts/dev-netns-shell.sh

I also added a really simple UDP test script that bounces a bunch of packets off of the board and counts how many got dropped:

https://github.com/alexforencich/verilog-ethernet/blob/master/scripts/udp_test.py

kimanha, 2020-09-08:
Thank you for adding nice script, udp_test.py is very useful for a quick test.

:~/tmp/verilog-ethernet/scripts$ python3 udp_test.py -n 100 10.0.142.128 1234
Sending 100 UDP packets to 10.0.142.128 on 1234...
Sent 100 packets
Received 100 packets (100.0%)
Missed 0 packets (0.0%)

By the way, can you show me an example how to use dev-netns-shell.sh ?

Thanks,

alexforencich, 2020-09-09:
run it with the name of the ethernet adapter that the FPGA is plugged in to.  For example, if you plugged the card into "eth0", then run:

sudo ./dev-netns-shell.sh eth0

This will create a new network namespace, put the adapter in the namespace, and start a shell in the namespace.  When you exit the shell, the namespace will be deleted.  From the shell, you can set an IP address on the interface and then use netcat, like so (note that the script exports the device name as $dev for convenience):

ip addr add 192.168.1.199/24 dev $dev
nc -u 192.168.1.128 1234

You can also run udp_test.py from the shell.  

The script is also smart enough that you can start several instances connected to the same ethernet interface, and the first one you start will set up the namespace while the last one you exit will tear down the namespace.

kimanha, 2020-09-10:
Hello,

I am using the Windows VM and Ubuntu 18 for my test setup, if I use dev-netns-shell.sh I can't communicate with target board.

Can you see test 2 udp_test.py? -n 100 no missing packets, but if I use -n 1000 I got a lots missing packets.
I tried less than 100 it works fine but more than 100, I saw packet missing.

Test 1:
$sudo ./dev-netns-shell.sh enp0s3
Creating network namespace 'enp0s3'
Adding interface 'enp0s3' to network namespace 'enp0s3'
Starting shell in network namespace 'enp0s3'
Note: $dev='enp0s3'

#ip addr add 10.0.142.199/24 dev $dev
#nc -u 10.0.142.128 1234
ddddd
ffffffffffffffff
^C

Test 2: without dev-netns-shell.sh run 

Connect Ethernet switch:

~/tmp/verilog-ethernet/scripts$ python3 udp_test.py -n 100 10.0.142.128 1234
Sending 100 UDP packets to 10.0.142.128 on 1234...
Sent 100 packets
Received 100 packets (100.0%)
Missed 0 packets (0.0%)

~/tmp/verilog-ethernet/scripts$ python3 udp_test.py -n 1000 10.0.142.128 1234
Sending 1000 UDP packets to 10.0.142.128 on 1234...
Sent 1000 packets
Received 244 packets (24.4%)
Missed 756 packets (75.6%)

Direct connection with PC.

~/tmp/verilog-ethernet/scripts$ python3 udp_test.py -n 100 10.0.142.128 1234
Sending 100 UDP packets to 10.0.142.128 on 1234...
Sent 100 packets
Received 100 packets (100.0%)
Missed 0 packets (0.0%)

~/tmp/verilog-ethernet/scripts$ python3 udp_test.py -n 1000 10.0.142.128 1234
Sending 1000 UDP packets to 10.0.142.128 on 1234...
Sent 1000 packets
Received 442 packets (44.2%)
Missed 558 packets (55.800000000000004%)

~/tmp/verilog-ethernet/scripts$ python3 udp_test.py -n 1000 10.0.142.128 1234
Sending 1000 UDP packets to 10.0.142.128 on 1234...
Sent 1000 packets
Received 215 packets (21.5%)
Missed 785 packets (78.5%)

Thanks,

alexforencich, 2020-09-10:
That's odd; not sure what's going on there.  Can you describe your test setup in more detail?  

With my laptop, if I connect the Arty direcly via a USB3 NIC, this is what I get:

    [alex@mizuchi scripts]$ sudo ./dev-netns-shell.sh enp0s20f0u1u2
    Creating network namespace 'enp0s20f0u1u2'
    Adding interface 'enp0s20f0u1u2' to network namespace 'enp0s20f0u1u2'
    Starting shell in network namespace 'enp0s20f0u1u2'
    Note: $dev='enp0s20f0u1u2'
    [root@mizuchi scripts]# ip addr add 192.168.1.199/24 dev $dev
    [root@mizuchi scripts]# ./udp_test.py 192.168.1.128 -n 100000
    Sending 100000 UDP packets to 192.168.1.128 on 1234...
    Sent 100000 packets
    Received 52977 packets (52.977%)
    Missed 47023 packets (47.022999999999996%)
    [root@mizuchi scripts]# ./udp_test.py 192.168.1.128 -n 100000
    Sending 100000 UDP packets to 192.168.1.128 on 1234...
    Sent 100000 packets
    Received 100000 packets (100.0%)
    Missed 0 packets (0.0%)
    [root@mizuchi scripts]# ./udp_test.py 192.168.1.128 -n 100000
    Sending 100000 UDP packets to 192.168.1.128 on 1234...
    Sent 100000 packets
    Received 100000 packets (100.0%)
    Missed 0 packets (0.0%)
    [root@mizuchi scripts]# exit
    exit
    Deleting network namespace 'enp0s20f0u1u2'
    [alex@mizuchi scripts]$ 

That's with 100000 packets per test.  The first test sees a bunch of drops due to ARP taking a while to respond to the FPGA, but subsequent tests show no packet loss.

kimanha, 2020-09-10:
Thank you for share your test result.
Your test result looks great.

My laptop is the Lenovo T580, onboard (Intel Ethernet Connection I219-V (Jacksonville)) port.

Wifi disabled.
Direct connection with Arty board.

Ethernet adapter Ethernet:
   Connection-specific DNS Suffix  . :
   IPv4 Address. . . . . . . . . . . : 10.0.142.100
   Subnet Mask . . . . . . . . . . . : 255.255.255.0
   Default Gateway . . . . . . . . . : 10.0.142.1

Arty IP config:
wire [31:0] local_ip          = {8'd10, 8'd0, 8'd142,   8'd128};
wire [31:0] gateway_ip    = {8'd10, 8'd0, 8'd142,   8'd1};
wire [31:0] subnet_mask = {8'd255, 8'd255, 8'd255, 8'd0};

I tried to run Windos-10 python 3.8.5.

Windows environment is much stable compare with Ubuntu environment .
1000 is stable, 5000 seems good but sometime got small errors, 100000 got ~ < 1% errors.

 

========================= RESTART: C:/Temp/udp_test.py =========================
Sending 1000 UDP packets to 10.0.142.128 on 1234...
Sent 1000 packets
Received 1000 packets (100.0%)
Missed 0 packets (0.0%)
>>> 
========================= RESTART: C:/Temp/udp_test.py =========================
Sending 1000 UDP packets to 10.0.142.128 on 1234...
Sent 1000 packets
Received 1000 packets (100.0%)
Missed 0 packets (0.0%)
>>> 
========================= RESTART: C:/Temp/udp_test.py =========================
Sending 1000 UDP packets to 10.0.142.128 on 1234...
Sent 1000 packets
Received 1000 packets (100.0%)
Missed 0 packets (0.0%)
>>> 
========================= RESTART: C:/Temp/udp_test.py =========================
Sending 1000 UDP packets to 10.0.142.128 on 1234...
Sent 1000 packets
Received 1000 packets (100.0%)
Missed 0 packets (0.0%)
>>> 
========================= RESTART: C:/Temp/udp_test.py =========================
Sending 1500 UDP packets to 10.0.142.128 on 1234...
Sent 1500 packets
Received 1500 packets (100.0%)
Missed 0 packets (0.0%)
>>> 
========================= RESTART: C:/Temp/udp_test.py =========================
Sending 1500 UDP packets to 10.0.142.128 on 1234...
Sent 1500 packets
Received 1500 packets (100.0%)
Missed 0 packets (0.0%)
>>> 
========================= RESTART: C:/Temp/udp_test.py =========================
Sending 5000 UDP packets to 10.0.142.128 on 1234...
Sent 5000 packets
Received 4882 packets (97.64%)
Missed 118 packets (2.36%)
>>> 
========================= RESTART: C:/Temp/udp_test.py =========================
Sending 5000 UDP packets to 10.0.142.128 on 1234...
Sent 5000 packets
Received 4864 packets (97.28%)
Missed 136 packets (2.7199999999999998%)

========================= RESTART: C:/Temp/udp_test.py =========================
Sending 5000 UDP packets to 10.0.142.128 on 1234...
Sent 5000 packets
Received 5000 packets (100.0%)
Missed 0 packets (0.0%)
>>> 
========================= RESTART: C:/Temp/udp_test.py =========================
Sending 5000 UDP packets to 10.0.142.128 on 1234...
Sent 5000 packets
Received 5000 packets (100.0%)
Missed 0 packets (0.0%)
>>> 
========================= RESTART: C:/Temp/udp_test.py =========================
Sending 5000 UDP packets to 10.0.142.128 on 1234...
Sent 5000 packets
Received 5000 packets (100.0%)
Missed 0 packets (0.0%)

========================= RESTART: C:/Temp/udp_test.py =========================
Sending 100000 UDP packets to 10.0.142.128 on 1234...
Sent 100000 packets
Received 99299 packets (99.299%)
Missed 701 packets (0.701%)
>>> 
========================= RESTART: C:/Temp/udp_test.py =========================
Sending 100000 UDP packets to 10.0.142.128 on 1234...
Sent 100000 packets
Received 99350 packets (99.35000000000001%)
Missed 650 packets (0.65%)
>>> 
========================= RESTART: C:/Temp/udp_test.py =========================
Sending 100000 UDP packets to 10.0.142.128 on 1234...
Sent 100000 packets
Received 99405 packets (99.405%)
Missed 595 packets (0.5950000000000001%)

alexforencich, 2020-09-10:
Windows host, with Ubuntu running in a VM?  Yeah, running the tests on the host OS will definitely be more reliable.  Not sure why you might be seeing drops; it's possible the test script isn't handling things very well---after all, it is incredibly simple.  I should probably rewrite it to use multithreading, that may improve things.

kimanha, 2020-09-10:
>>Windows host, with Ubuntu running in a VM? 
Yes, exactly!
>>I should probably rewrite it to use multithreading
Sounds great, FPFA UDP MTU size is 1500 byte or jumbo frame support?

Next, my target board is the Xilinx ZCU102 board.
Do you have any update status for the ZCU102 board?
I think your ZCU106 project possibly converter to the ZCU102 board.
I already converted the XDC file and PLL clock and GTH port for the ZCU102 board(ready to test)

I am looking 10 G Ethernet adaptor for 10G project test.
Do you think this 10G adaptor is working with your UDP project or any recommendation?
https://www.sonnettech.com/product/solo10g-tb3/overview.html

Thank you for your support.

alexforencich, 2020-09-10:
I think the only limit is the FIFO sizes.  So you may need to adjust the FIFO sizes for things to work with 9 KB MTU frames.  

I do not own a ZCU102 board, so I cannot test the design on that board.  Presumably porting from the ZCU106 should be straightforward.  

That adapter is probably fine, but the ZCU102 only has SFP+ cages, so you'll also need to get an SFP+ to 10GBASE-T adapter.  My recommendation would be to get an old desktop PC, install a PCIe 10G NIC, and pick up an SFP+ DAC cable.

kimanha, 2020-09-10:
I copied from issue #16 your answer.
>>Also, my standing offer for anyone who wants a design ported to a new board: donate a copy of the hardware, and I'll port it and maintain it.

This offer is still valid if I donate the ZCU102 board?

Thanks,

alexforencich, 2020-09-10:
Like I said, it's a standing offer that applies to all of my projects.  However, I don't make any promises on timelines, although I will prioritize it as high as I reasonably can.  For porting verilog-ethernet to ZCU102, that should be no problem at all and I should be able to get that turned around quickly.

alexforencich, 2020-09-10:
Sure.  But I'll need your email address to do that.  Either you can post it, or you can send me an email at alex at alexforencich dot com.

closed 2020-09-13 by kimanha