alexforencich/verilog-ethernet #43: missing parts for PTP and 1g example design with PTP source

gruberth96, 2020-09-23:
First of all thank you for your excellent work regarding the verilog ethernet and axi repos.
I am currently trying to understand and activate your PTP components and i am uncertain because i read in a different issue that PTP is not necessarily 100% complete. What's here exactly missing? And is there an example where PTP is activated and PTP source clock is connected?

I already found the three parameters in the eth_mac_phy_10g_fifo file to activate PTP for 10G design, it synthesize and generate+connect all PTP related components.
Is it possible to copy it simple to the 1G design with the 10G design as "template"? Or is it necessary to pay attention at some stages?

alexforencich, 2020-09-23:
The only PTP components in this repo are the PTP clock, PTP CDC module, and PTP timestamping support in the MACs.  There is no implementation of the PTP protocol itself, time servo, master clock selection, etc.  The components were added in support of http://github.com/corundum/corundum, where the PTP protocol is implemented with linuxptp.  The only reason the incomplete PTP code is in this repo at the moment is to support corundum, otherwise I would not have committed it in its current state.  There is also no standalone PTP implementation at the moment, and I am not currently planning on writing one.  As a result, there are also no example designs for PTP in this repo, but corundum can serve as an example design.  

Yes, you can copy over the 10G MAC code for PTP to the gigabit MACs.  However, the FIFO modules will get a LOT larger as they will store 96 extra bits for every byte in the FIFO to hold the PTP timestamp.  On my to-do list are reworking the PTP CDC logic to make it a bit more efficient and improve timing performance and reworking the FIFO modules to handle metadata more effectively.  Once those are done, then I will most likely add PTP support to the other MACs as well.  Also, I will probably do away with the extra PTP timestamp ports and keep the PTP timestamps in the tuser sideband signal.

gruberth96, 2020-09-24:
Okay thanks for your quick response. Yes i saw that and didn't expect that at all a full stack in HW. I only need timestamping of the RX and TX packtes. The PTP protocol stack implementation will be done in SW with linuxptp or ptpd. And i also don't need a PTP standalone version, i will use for sure your ethernet MAC therefore it's fine. 

Ok thank you for your hints.

closed 2020-09-24 by gruberth96

gruberth96, 2020-12-04:
Hi Alex,
what is the purpose of the 16 bit wide PTP Tag? 
What should be the input tag for the fifo module and what should be done afterwards in the SW stack because it's saved with the tx timestamp in the tx_ptp_ts fifo module?

And how do you select if you save a timestamp for a TX message? Or is for every TX message a timestamp saved, same as for RX side? First i thought this is done via the tag input, but without setting the tag_valid signal the loopback won't work i detected.

Best regards

alexforencich, 2020-12-04:
The idea with the tag is that you can attach it to the TX frame, and it will be returned attached to the timestamp.  So you can use it to track which timestamps were associated with which packets.  All TX packets are time stamped, same as on the RX side.

I'm actually planning on reworking the PTP interface at some point and rolling the timestamp and tag into tuser.  However, I need to write some new FIFO modules before I do that so the FIFOs don't take up an unreasonable amount of block RAM resources.

gruberth96, 2020-12-04:
Okay thanks for the info. This means a good approach would be to add the sequenceId which is generated by the SW stack. And if the transmitted message is no ptp relevant one to just set it default to zero. Because for non ptp relevant messages you don't need the timestamp at all if i understood right. 
Or how do you use it in corundum with combination to linuxptp?

alexforencich, 2020-12-04:
When PTP timestamping support is enabled, Corundum unconditionally returns RX and TX timestamps for all packets in the completion records, similar to many other NICs.  The tag field is currently not used, but I will probably look in to implementing that at some point, at least to make things a bit more robust on the transmit path.