alexforencich/verilog-ethernet #44: DE2115 TXD

nitya261, 2020-09-23:
Hi, 

I'm running the ethernet code provided for DE2-115. I'm able to see data from AXI input, AXI output signals and  rgmii_rxd till eth_mac_1g_rgmii_fifo.v but not rgmii_txd. Also ENET0_RX_DATA, ENET0_RX_DV but not ENET0_TX_DATA, ENET0_TX_EN. I cannot add signals from "oddr" to Signal Tap Logic Analyzer in Quartus. Can you help me out in finding the issue. TIA

alexforencich, 2020-09-23:
Most likely you cannot monitor signals from within the IOBs with signaltap.  Closest that I suspect you'll be able to get is the internal GMII interface between the MAC and the DDR flip-flops in the IOBs.

nitya261, 2020-09-24:
So I can see data for txd and rxd before I can connect to phy(oddr) i.e., after eth_mac.  I can see only rxd thereafter but not txd as I mentioned before. What could be the issue or how can I test this?

I'm unable to add internal GMII signals (eth_mac_1g_rgmii.v signals) too. I can only add  ENET0_TX_DATA from fpga.v and gmii_txd from eth_mac_1g.v. I see data from gmii_txd but not from ENET0_TX_DATA.

nitya261, 2020-09-29:
Hi I'm doing this in windows. Also compiled in linux with provided makefile but still I can't see data after ddr/final output. Able to see it before oddr as in windows. Just wanted to know if this tested on DE2-115 board? 
Thank you

closed 2020-09-30 by nitya261