alexforencich/verilog-ethernet #45: Making of Example Vivado Project

FPGA-Bot-Yang, 2020-09-23:
Hi Alex,

Hope you are doing great!

When I trying to make the example Vivado Project under /example/VCU118/fpga_10g, I run into the following error message:

```
cd fpga && make
make[1] : Entering directory 'xxxx/example/VCU118/fpga_10g/fpga'
make[1] : *** NO rule to maek target 'fpga.bit', need by 'fpga'. Stop.
make[1]: Leaving directory 'xxxx/example/VCU118/fpga_10g/fpga'
Makefile: 14: recipe for target 'fpga' failed
make: *** [fpga] Error 2
```

I've already added the binary file path of Vivado 2019.1 to PATH.

I think this should be a simple fix. Could you give me a hint on what I might missing in my configuration?

Thank you!

alexforencich, 2020-09-23:
That's the very useful error message that make spits out when it can't find a source file.  Are you running on linux or windows?  Windows doesn't support symlinks, which are used in this repo to point to the main verilog files.  If that symlink isn't there, then the files will not be found.  I suggest editing the file list in the makefile to point to the files directly, instead of relying on the symlink in the lib directory.

FPGA-Bot-Yang, 2020-09-24:
I've replaced each file included in the Makefile under `/example/VCU118/fpga_10g/fpga`, but still having the same issue. Is there any other files I should modify in this case?

Thanks!

closed 2020-09-24 by FPGA-Bot-Yang

alexforencich, 2020-09-24:
Ah, good to hear!  I need to do some more digging to see if there is a way to get make to be a little bit more verbose when it can't find a file, as that would make solving this sort of issue much easier.

FPGA-Bot-Yang, 2020-09-24:
> Ah, good to hear! I need to do some more digging to see if there is a way to get make to be a little bit more verbose when it can't find a file, as that would make solving this sort of issue much easier.

Absolutely looking forward to that. :-) A similar issue is existing in the PCIe design of yours. I wrote a bash script to regenerate the symbolic link of all the affected links.

IncognitoChap, 2021-11-03:
> I've replaced each file included in the Makefile under `/example/VCU118/fpga_10g/fpga`, but still having the same issue. Is there any other files I should modify in this case?
> 
> Update: I fixed it by recreating the symbolic link under `/example/VCU118/fpga_10g/lib` and `/example/VCU118/fpga_10g/tb`.
> 
> Thanks for all the help!

Hi there, I am also not able to get it working with similar error faced by you. However, I do not understand how you fixed it (with the recreating symbolic link method) Do you mind elaborating it? Thanks in advance!

alexforencich, 2021-11-03:
If you're building on Linux, it should Just Work (tm).  Building on windows with the makefiles is not really supported since windows filesystems do not support symlinks, and I make heavy use of symlinks to reduce the use of relative paths that go up many directory levels.  

Most of the symlinks in this repo are just a link back to the repo root.  If you deleted a symlink accidentally, you can check it out again with `git checkout <file>`.  Or you cou can recreate it by CDing into the 'lib' directory and running `ln -s ../../../../ eth`.  This will not work in Windows, in that case you'll just have to manually create the vivado project, add all of the verilog files and constraints files, and source all of the tcl scripts.  All of the files you need are listed in the makefile.

IncognitoChap, 2021-11-03:
Thanks Alex for the prompt response. I am still contemplating if I want to do all this in Linux. But point taken and it seems like I will do everything manually for now in Windows .