alexforencich/verilog-ethernet #46: eth_mac_mii_fifo module not fully working for me

MichaelAstahov, 2020-09-24:
Hello. First of all, thanks for this great ethernet IP.

I'm trying to use this ethernet mac & UDP/IP stack for my project, my board is Lattice ECP3 Versa, Marvel 88E1111 PHY.
I took the Arty example as a reference for implementing the ethernet MII & UDP/IP interface to my FPGA.

I did as in the example and instantiated fpga_core module like this:
```
ethernet_core ethernet_inst (
    .clk		 (phy_clk125),
    .rst		 (rst),
	.led		 (led),
 
    .phy_rx_clk	 (phy_rx_clk),
    .phy_rxd	 (phy_rxd[3:0]),
    .phy_rx_dv	 (phy_rx_dv),
    .phy_rx_er	 (phy_rx_er),
    .phy_tx_clk	 (phy_tx_clk),
    .phy_txd	 (phy_txd[3:0]),
    .phy_tx_en	 (phy_tx_en),
    .phy_col	 (phy_col),
    .phy_crs	 (phy_crs),
    .phy_rst_n   (phy_rst_n)
);
```
I just made a few changes that don't make any difference like delete the UART signals & Switches I don't use.
In the example, you created the 125MHz clock from Xilinx PLL, in my case I used the Marvel PHY 125MHz clock,
from Marvel 88E1111 PHY datasheet:
[image]

Next, inside the fpga_core, where the eth_mac_mii_fifo module instantiated I make sure that the TARGET parameters is "GENERIC":
```
eth_mac_mii_fifo #(
    .TARGET("GENERIC"),
    .CLOCK_INPUT_STYLE("BUFR"),
    .ENABLE_PADDING(1),
    .MIN_FRAME_LENGTH(64),
    .TX_FIFO_DEPTH(4096),
    .TX_FRAME_FIFO(1),
    .RX_FIFO_DEPTH(4096),
    .RX_FRAME_FIFO(1)
)
eth_mac_inst (
    .rst(rst),
    .logic_clk(clk),
    .logic_rst(rst),

    .tx_axis_tdata(tx_axis_tdata),
    .tx_axis_tvalid(tx_axis_tvalid),
    .tx_axis_tready(tx_axis_tready),
    .tx_axis_tlast(tx_axis_tlast),
    .tx_axis_tuser(tx_axis_tuser),

    .rx_axis_tdata(rx_axis_tdata),
    .rx_axis_tvalid(rx_axis_tvalid),
    .rx_axis_tready(rx_axis_tready),
    .rx_axis_tlast(rx_axis_tlast),
    .rx_axis_tuser(rx_axis_tuser),

    .mii_rx_clk(phy_rx_clk),
    .mii_rxd(phy_rxd),
    .mii_rx_dv(phy_rx_dv),
    .mii_rx_er(phy_rx_er),
    .mii_tx_clk(phy_tx_clk),
    .mii_txd(phy_txd),
    .mii_tx_en(phy_tx_en),
    .mii_tx_er(),

    .tx_fifo_overflow(),
    .tx_fifo_bad_frame(),
    .tx_fifo_good_frame(),
[... 625 lines cut ...]
    .s_axis_tready(rx_fifo_udp_payload_axis_tready),
    .s_axis_tlast(rx_fifo_udp_payload_axis_tlast),
    .s_axis_tid(0),
    .s_axis_tdest(0),
    .s_axis_tuser(rx_fifo_udp_payload_axis_tuser),

    // AXI output
    .m_axis_tdata(tx_fifo_udp_payload_axis_tdata),
    .m_axis_tkeep(),
    .m_axis_tvalid(tx_fifo_udp_payload_axis_tvalid),
    .m_axis_tready(tx_fifo_udp_payload_axis_tready),
    .m_axis_tlast(tx_fifo_udp_payload_axis_tlast),
    .m_axis_tid(),
    .m_axis_tdest(),
    .m_axis_tuser(tx_fifo_udp_payload_axis_tuser),

    // Status
    .status_overflow(),
    .status_bad_frame(),
    .status_good_frame()
);

endmodule
	
```

alexforencich, 2020-09-24:
Looks like m_axis_tuser was asserted with m_axis_tlast, so the FIFO dropped the frame.  

What toolchain are you using?  It's possible that it doesn't like the lfsr code, and hence the fcs check is failing.

MichaelAstahov, 2020-09-24:
> Looks like m_axis_tuser was asserted with m_axis_tlast, so the FIFO dropped the frame.
> 
> What toolchain are you using? It's possible that it doesn't like the lfsr code, and hence the fcs check is failing.

oh, I didn't realize m_axis_tuser is a signal the shouldn't be asserted.
Here is an analyzer of the whole axis_gmii_rx module, including the inputs/outputs from LFSR module:
[image]

yes, there is bad_frame asserted. why it happening? what I doing wrong?

the toolchain I'm using is just Diamond software from Lattice to program the FPGA (Lattice ECP3 Versa Board) and Reveal Analyzer to debug the logic inside (the pictures I attached).

alexforencich, 2020-09-24:
Try swapping out the lfsr module instance in both of the axis_gmii modules for this statically-generated file:

https://github.com/alexforencich/verilog-ethernet/blob/47ca9a8725dba00990341470b6af37148c50c5ec/rtl/eth_crc_8.v

and let me know what happens.  Some crappy tools do not perform constant elaboration correctly, and hence the parametrized lfsr module does not work.

MichaelAstahov, 2020-09-24:
> Try swapping out the lfsr module instance in both of the axis_gmii modules for this statically-generated file: [...]

yes, that worked! thank you!!
Now I'm getting the Rx from RJ45 to the eth_mac_mii_fifo module output as expected.
I will investigate the code further now, if I will have another question I open another issue here.
Thanks.

closed 2020-09-24 by MichaelAstahov