alexforencich/verilog-ethernet #47: Unable to connect MII Ethernet interface with 1G/100M/10M switch

MichaelAstahov, 2020-09-30:
Hello, I added the MII interface as shown in the [Arty example](https://github.com/alexforencich/verilog-ethernet/tree/master/example/Arty/fpga) to my FPGA (I'm using Lattice ECP3 Dev Board with Marvel PHY 88e1111).
Although my PHY/Cables and Switches support 1G (GMII) I prefer to work with MII because some of my equipment doesn't support 1G.

The ethernet interface is working when I connect the FPGA through a switch that supports 10/100 only.
If I connect the FPGA through a switch that supports 10/100/1G I getting 8bit of packets to my FPGA (I connected all the 8bits to the FPGA so I can see what happens although I need to use only the 4 LSB in the MII interface), and the MII interface doesn't do anything with this data (I anyway connecting 4LSB to the ethernet core).

**Case 1: 10/100 Switch -> MII Interface**
[image]
[image]
**as can be seen in this case the FPGA receives only 4bits and the MAC successfully processing this data.**

--

**Case 2: 10/100/1G Switch -> MII Interface**
[image]
[image]
**as can be seen in this case the FPGA receives 8bits and the MAC doesn't do anything with the data.**

-----------------------------------------------------------------------------------------

**My question is how I can adjust the MII ethernet interface to work with 10/100/1G switches and cables?
the interface doesn't need to notify the LAN about that he can receive only 10/100 data packets? 
I have to implement a 3-state mac interface or there is a way using the mii interface with a 3-state switch?**

-------------------------------------------------------------------------------------------

 **top level code:**

```
module top (
    // board clock&reset / leds&switches
    input               clk_board,      // 100MHz versa board clock    	
    input               reset_n,        // rst neg - push button       
    output	[7:0]  		led,            // led array                                           
    output  		    dp,             // 7-seg screen led                               
    input 	[7:0]    	sw,             // switches                                

    // Ethernet: 100BASE-T MII   
    input               phy_tx_clk,                                                                                    
    output [7:0]        phy_txd,                                                                                   
    output              phy_tx_en,                                 
                                                                              
    input               phy_rx_clk,                                                       
    input  [7:0]        phy_rxd,       
    input               phy_rx_dv,                
    input               phy_rx_er,  
	
    input               phy_col,                                                                                 
    input               phy_crs,  
    output              phy_mdc,                                                                       
    inout               phy_mdio,                                      
	
	input               phy_clk125,
    output              phy_rst_n
	);

// ------- rst_n & rst -------- //	
wire rst, rst_n;
reg [7:0] count; 
assign rst_n = (count > 8'b1111_1010) ? 1'b1 : 1'b0;
assign rst = !rst_n;
		
always @(posedge clk_100 or negedge reset_n) begin
	if (~reset_n)
		count = 8'b0000_0000;
	else if (count < 8'b1111_1101)
		count = count + 1;
end	

CLKDIVB clkdiv (
[... 544 lines cut ...]
    .s_axis_tvalid(rx_fifo_udp_payload_axis_tvalid),
    .s_axis_tready(rx_fifo_udp_payload_axis_tready),
    .s_axis_tlast(rx_fifo_udp_payload_axis_tlast),
    .s_axis_tid(0),
    .s_axis_tdest(0),
    .s_axis_tuser(rx_fifo_udp_payload_axis_tuser),

    // AXI output
    .m_axis_tdata(tx_fifo_udp_payload_axis_tdata),
    .m_axis_tkeep(),
    .m_axis_tvalid(tx_fifo_udp_payload_axis_tvalid),
    .m_axis_tready(tx_fifo_udp_payload_axis_tready),
    .m_axis_tlast(tx_fifo_udp_payload_axis_tlast),
    .m_axis_tid(),
    .m_axis_tdest(),
    .m_axis_tuser(tx_fifo_udp_payload_axis_tuser),

    // Status
    .status_overflow(),
    .status_bad_frame(),
    .status_good_frame()
);

endmodule
```

alexforencich, 2020-09-30:
If you have a GMII PHY and don't prevent the PHY from negotiating to 1 Gbps, then have to use a GMII MAC.  Basically, the PHY will use GMII when running at 1 Gbps or MII when running at 10 or 100 Mbps.  The MAC has no control over this at all.  If you want to force a 1 Gbps PHY to run at 10/100 Mbps, then you'll have to poke some registers over MDIO.  No idea which ones offhand, consult the PHY datasheet.  Take a look at the VCU118 design for some MDIO init code that you might be able to adapt.

MichaelAstahov, 2020-10-01:
> If you have a GMII PHY and don't prevent the PHY from negotiating to 1 Gbps, then have to use a GMII MAC. Basically, the PHY will use GMII when running at 1 Gbps or MII when running at 10 or 100 [...]

ok so I decided to try GMII, I did it as in the KC705 example.
the problem is I get synthesis error in the oddr.v block:

```
    reg [WIDTH-1:0] d_reg_1 = {WIDTH{1'b0}};
    reg [WIDTH-1:0] d_reg_2 = {WIDTH{1'b0}};
    reg [WIDTH-1:0] q_reg    = {WIDTH{1'b0}}; 
    
    always @(posedge clk) begin
        d_reg_1 <= d1;
        d_reg_2 <= d2;
    end

     always @(posedge clk) begin
        q_reg <= d1;
    end

    always @(negedge clk) begin
        q_reg <= d_reg_2;
    end

    assign q = q_reg;
```

my synthesis tool doesn't support this coding style (Synplify Pro with Lattice FPGA, Diamond software)
the error I get is that I can't assign to the same register in two different always blocks.

One of the solutions I tried is to use ODDRX block from Lattice:
[image]
but this also makes me a lot of problems because it says the 'SCLK' cant be a general-purpose clock and I can't get rid of this error from Lattice (I tried to make this clock a Primary Clock but this didn't solve the problem - I hate Lattice tools..).

Another solution I tried is to use another ODDR block I found:

```
module oddrx
(
// Primary I/O 
DA, DB, RST, CLK_I, CLK_IX2, Q
);

input DA; 
input DB; 
input CLK_I; 
input CLK_IX2; 
input RST; 
output Q;

reg  DA_f;
reg  DB_f;
reg  Q;
wire MUX_OUT;

   assign MUX_OUT = CLK_I ? DA_f:DB_f;

   always @(posedge CLK_I or posedge RST) begin
	  if (RST) begin
             DA_f <= 1'b0;
             DB_f <= 1'b0;
	  end
	  else begin
             DA_f <= DA;
             DB_f <= DB;
	  end
   end

   always @(negedge CLK_IX2 or posedge RST) begin
	  if (RST) begin
             Q <= 1'b0;
	  end
	  else begin
             Q <= MUX_OUT;
	  end
   end

endmodule
```

and Instantiated this block as follows:

```
module ssio_sdr_out #
(
    // target ("SIM", "GENERIC", "XILINX", "ALTERA")
    parameter TARGET = "GENERIC",
    // IODDR style ("IODDR", "IODDR2")
    // Use IODDR for Virtex-4, Virtex-5, Virtex-6, 7 Series, Ultrascale
    // Use IODDR2 for Spartan-6
    parameter IODDR_STYLE = "IODDR2",
    // Width of register in bits
    parameter WIDTH = 1
)
(
    input  wire             clk,                // mac_tx_clk
    input  wire [WIDTH-1:0] input_d,            // input_data -> mac data

    output wire             output_clk,         // phy_tx_clk
    output wire [WIDTH-1:0] output_q            // output_data -> phy data
);

/* oddr #(
    .TARGET(TARGET),
    .IODDR_STYLE(IODDR_STYLE),
    .WIDTH(1)
)
clk_oddr_inst (
    .clk(clk),
    .d1(1'b0),
    .d2(1'b1),
    .q(output_clk)
); */

oddrx oddrx_clk_inst (
    .DA         (1'b1), 
    .DB         (1'b0), 
    .RST        (1'b0),
    .CLK_I      (clk), 
    .CLK_IX2   (~clk), 
    .Q          (output_clk)
);

(* IOB = "TRUE" *)
reg [WIDTH-1:0] output_q_reg = {WIDTH{1'b0}};

assign output_q = output_q_reg;

always @(posedge clk) begin
    output_q_reg <= input_d;
end

endmodule
```

this is also doesn't work for me (looks like gtx_clk not ticking.. that means this DDR block or how I instantiated it is not good).
Do you have any solution up your slave? I will be really glad.

by the way.. the FPGA using the GMII interface does catch the signals now from the 10/100/1G switch and process it, but doesn't respond to ARP requests, so it's probably a problem in the DDR output, right?

Thanks!

alexforencich, 2020-10-01:
Yeah, that oddr code is for simulation only, you will need to use the correct primitive for your device.  Generally I don't think you can infer DDR inputs and outputs reliably.  I don't have access to any lattice parts, so they are not currently supported, you'll have to figure that part out.  

Yeah, if the output isn't working, then it won't be able to respond.

MichaelAstahov, 2020-10-01:
> Yeah, that oddr code is for simulation only, you will need to use the correct primitive for your device. Generally I don't think you can infer DDR inputs and outputs reliably. I don't have access to [...]

ok ill try to instantiate the oddr block from lattice. thanks.

closed 2020-10-01 by MichaelAstahov