alexforencich/verilog-ethernet #48: respond to ARP through direct connection but doesn't respond through the LAN

MichaelAstahov, 2020-10-01:
Hi.
I added your GMII & UDP stack like the KC705 example.

when I connecting the Ethernet cable from the FPGA to the PC, there is not a lot of traffic, {~1packet / 1sec on average},
I make ARP broadcast from my computer and the FPGA responding correctly and I can see the FPGA on the LAN (arp -a).

On the other hand, when I connect the FPGA to my LAN switch, {~20packets / 1sec on average}, and the PC also connected to this switch, once again I broadcast ARP requests through the LAN, I can see this request on the LAN using Wireshark, but the FPGA not responding to it.

Any idea why that might happen? Does the FPGA miss this request because of a heavy load of traffic on the cable? seems unreasonable, any idea what going on?

Thanks!
(btw my solution to this issue: https://github.com/alexforencich/verilog-ethernet/issues/47 was to pass the input clock in Lattice PLL and output 125MHz clock... I think it's a good enough solution for now no?)

MichaelAstahov, 2020-10-07:
https://electronics.stackexchange.com/questions/524751/ethernet-communication-with-udp-ip-stack-working-with-direct-connection-but-does
working now with your help in Stack Exchange, thanks.

closed 2020-10-07 by MichaelAstahov

alexforencich, 2020-10-07:
Oh shoot, somehow I did not get notified about this issue at all.  I was wondering why you asked about it on stack exchange instead of on here....sigh....

zaidtahirbutt, 2022-06-22:
> https://electronics.stackexchange.com/questions/524751/ethernet-communication-with-udp-ip-stack-working-with-direct-connection-but-does working now with your help in Stack Exchange, thanks.

@MichaelAstahov What was the solution? The stackexchange link has been removed.

alexforencich, 2022-06-22:
The solution is that the lattice tools suck and don't properly handle my parametrizable LFSR module, so the Ethernet FCS computation module instances had to be swapped out for the old generated CRC computation modules.

zaidtahirbutt, 2022-06-22:
> On the other hand, when I connect the FPGA to my LAN switch, {~20packets / 1sec on average}, and the PC also connected to this switch, once again I broadcast ARP requests through the LAN, I can see this request on the LAN using Wireshark, but the FPGA not responding to it.

@alexforencich so once the old generated CRC computation modules are inserted, then the FPGA is able to communicate with the host through the LAN switch (as highlighted in the question from this thread I quoted above)?