alexforencich/verilog-ethernet #49: subnet decision issue

Rhapso, 2020-10-05:
```verilog
if (((arp_request_ip ^ gateway_ip) & subnet_mask) == 0) begin
                // within subnet, look up IP directly
                // (no bits differ between request IP and gateway IP where subnet mask is set)
      // some logic here
end
```
Is the expression to decide whether within subnet correct? I am confused.

Rhapso, 2020-10-05:
PS. These codes are in `rtl/arp.v`.

alexforencich, 2020-11-28:
Yes, this is a bit-masked comparison.  XOR returns 1 for each bit position the request IP differs from the gateway IP, and then this is masked off by the subnet mask as we only want to know if the address is within the subnet or not.  If the result is 0, then it's the same subnet.  If the result is not zero, then there is at least one bit different, indicating that the address is in a different subnet.