alexforencich/verilog-ethernet #53: support 40G

123wangke, 2020-11-21:
Hello, may I ask you a question? 40g is composed of four 10g channels. Referring to 802.3, we know that each channel adopts 64b / 66b coding standard. The axis stream output from Xilinx's MAC / PHY IP core is 256 bit wide. Where is the data merging of the four channels completed? Is it at the interface between the physical layer and the MAC layer

alexforencich, 2020-11-21:
The merging is done in the PCS layer.  See 802.3 section 82.  It is far more complicated than 10G Ethernet.  40G Ethernet uses something called XLGMII between the MAC and the PHY, which is very similar to XGMII, except the packet start location is more restrictive, using 8 byte alignment instead of 4 byte alignment (64b/66b block types 2d, 33, 66 and 55 are not allowed) and the interface can be N*64 bits wide (incidentally, CGMII for 100G ethernet is exactly the same as XLGMII).  I think 256 bit wide XLGMII is common.  Then, the PCS layer handles striping the data across the four virtual lanes, inserting alignment markers, and transmitting the data.  On the receive side, each physical lane has to detect the alignment markers, then they get remapped to the correct virtual lanes.  Each lane has a FIFO that is used for deskew and there is a deskew routine to get that sorted out so all of the lanes are aligned.

123wangke, 2020-11-25:
Thank you for your answer. I've seen the simulation file before. I'm still confused about the setting of the minimum 16 byte frame spacing. I wonder why you didn't set it to 12 bytes. Now I understand. Thank you again

alexforencich, 2020-11-28:
Which 16 byte setting?

JakaBac, 2024-09-11:
There seems to be an Apache 2.0 licensed 100G/50G/40G/25G/10Gbps MAC that was tested on VC709 development board.
https://github.com/lewiz-support/LMAC_CORE3