alexforencich/verilog-ethernet #55: tx parameters

nitya261, 2020-11-25:
Hi, I'm trying to modify the code. If not for loopback, how can determine values for below parametrs in fpga_core module:

tx_udp_hdr_valid
rx_udp_hdr_ready
tx_udp_length

alexforencich, 2020-11-28:
If you have the checksum module enabled, then tx_udp_length can be set to 0.  As far as the udp_hdr_valid/ready signals, these need to be driven from some sort of a state machine to source/sink the headers associated with each packet - you should have one transfer cycle on udp_hdr for every frame (delineated by tlast) on udp_payload.  One of these days, I need to add some code to the example designs for how to send packets in addition to the loopback path.

nitya261, 2020-11-30:
Thanks Alex. How about tuser?

alexforencich, 2020-12-01:
tuser[0] is used to indicate bad frames.  Tie it low, unless you want to mark a frame as bad, which will then be dropped by the TX FIFO.

nitya261, 2020-12-02:
Thanks Alex

closed 2020-12-07 by nitya261