alexforencich/verilog-ethernet #58: lfsr module (eth_crc_8 instance) in the axis_gmii_rx module is not synthesizing correctly with Precision and Synplify Pro

tkunce, 2020-12-01:
The LFSR module (eth_crc_8 instance) is producing warnings and not synthesizing correctly in both Mentor Precision and Synopsys Synplify Pro tools. Both tools in the RTL view show that data_in is not connected to anything, and there are multiple warnings in the log files stating that only specific constructs can be accepted. 

From Precision's log:
# Warning: [45703]: "../in/hdl/lfsr.v", line 274: This statement has been ignored in the initial block. Possible reason is nonstatic initialization or assignment to a memory signal. 

From Synplify Pro's log:
@W: CG532 :"../in/hdl/lfsr.v":211:0:211:6|Within an initial block, only Verilog force statements and memory $readmemh/$readmemb initialization statements are recognized, and all other content is ignored.

This ends up causing the FCS to fail at the end of a packet, and in turn cause all received packets to get discarded. I recommend re-coding this module to avoid the use of the initial blocks so that any synthesis tool can synthesize this module. If there is a workaround then that should be published in the readme. 

Setup: Xilinx XC5VLX330T-1FFG1738C, Precision 2020.1, Synplify Pro P-2019.09, 1 Gb Ethernet MAC (SGMII) with UDP/IPv4 stack design

alexforencich, 2020-12-01:
This issue has been reported before.  The workaround for broken synthesizers is to pull the old non-parametrized code from the git history.  I will rewrite the module when I have time, but it is not a high priority as it works fine with all of the tools I have access to.

tkunce, 2020-12-01:
I appreciate the quick feedback. I will try pulling the old code and post the steps to getting it fixed here after I verify it in hardware. Thank you

alexforencich, 2020-12-01:
https://github.com/alexforencich/verilog-ethernet/blob/47ca9a8725dba00990341470b6af37148c50c5ec/rtl/eth_crc_8.v .  If you need the other widths, they should be in the history at the same point.

tkunce, 2020-12-01:
I have yet to look if the other LFSR instances are synthesized incorrectly, but I will use that time frame of the file you posted to search around for all the other LFSR modules if need be. Hopefully will have everything working tomorrow and I will note each file I needed to change here. Thanks again.

tkunce, 2020-12-02:
There were a total of 4 instances which need to be replaced when using the 1 Gb Ethernet MAC and UDP/IPv4 stack. After replacing them I was able to get traffic in both directions on my hardware. I will show how to replace them below for anyone else experiencing this problem:

----------------------------------------------------------------------------------------
axis_gmii_rx.v -> eth_crc_8: Replace with lfsr with the eth_crc_8 module
----------------------------------------------------------------------------------------

//lfsr #(
//    .LFSR_WIDTH(32),
//    .LFSR_POLY(32'h4c11db7),
//    .LFSR_CONFIG("GALOIS"),
//    .LFSR_FEED_FORWARD(0),
//    .REVERSE(1),
//    .DATA_WIDTH(8),
//    .STYLE("AUTO")
//)
//eth_crc_8 (
//    .data_in(gmii_rxd_d4),
//    .state_in(crc_state),
//    .data_out(),
//    .state_out(crc_next)
//);

// Old CRC logic used to get synthesis working
eth_crc_8
eth_crc_8_inst (
    .data_in(gmii_rxd_d4),
    .crc_state(crc_state),
    .crc_next(crc_next)
);

----------------------------------------------------------------------------------------
axis_gmii_tx.v -> eth_crc_8: Replace with lfsr with the eth_crc_8 module
----------------------------------------------------------------------------------------

//lfsr #(
//    .LFSR_WIDTH(32),
//    .LFSR_POLY(32'h4c11db7),
//    .LFSR_CONFIG("GALOIS"),
//    .LFSR_FEED_FORWARD(0),
//    .REVERSE(1),
//    .DATA_WIDTH(8),
//    .STYLE("AUTO")
//)
//eth_crc_8 (
//    .data_in(s_tdata_reg),
//    .state_in(crc_state),
//    .data_out(),
//    .state_out(crc_next)
//);

// Old CRC logic used to get synthesis working
eth_crc_8
eth_crc_8_inst (
    .data_in(s_tdata_reg),
    .crc_state(crc_state),
    .crc_next(crc_next)
);

----------------------------------------------------------------------------------------
arp_cache.v -> rd_hash and wr_hash: Replace with lfsr with the eth_crc_32 module
----------------------------------------------------------------------------------------

//lfsr #(
//    .LFSR_WIDTH(32),
//    .LFSR_POLY(32'h4c11db7),
//    .LFSR_CONFIG("GALOIS"),
//    .LFSR_FEED_FORWARD(0),
//    .REVERSE(1),
//    .DATA_WIDTH(32),
//    .STYLE("AUTO")
//)
//rd_hash (
//    .data_in(query_request_ip),
//    .state_in(32'hffffffff),
//    .data_out(),
//    .state_out(query_request_hash)
//);

// Old CRC logic used to get synthesis working
eth_crc_32
rd_hash (
    .data_in(query_request_ip),
    .crc_state(32'hffffffff),
    .crc_next(query_request_hash)
);

//lfsr #(
//    .LFSR_WIDTH(32),
//    .LFSR_POLY(32'h4c11db7),
//    .LFSR_CONFIG("GALOIS"),
//    .LFSR_FEED_FORWARD(0),
//    .REVERSE(1),
//    .DATA_WIDTH(32),
//    .STYLE("AUTO")
//)
//wr_hash (
//    .data_in(write_request_ip),
//    .state_in(32'hffffffff),
//    .data_out(),
//    .state_out(write_request_hash)
//);

// Old CRC logic used to get synthesis working
eth_crc_32
wr_hash (
    .data_in(write_request_ip),
    .crc_state(32'hffffffff),
    .crc_next(write_request_hash)
);