alexforencich/verilog-ethernet #59: How to make this project for C10LP

dennypig, 2020-12-10:
Hello author,
I have some problem with make,
"Altera Quartus toolchain components are in PATH"
What does it mean?
I git the full project from your github, and cd to verilog-ethernet/example/C10LP/
and run make to try to build,
but some error happened.

ee420@ee420-VirtualBox:~/Desktop/verilog-ethernet-master/example/C10LP/fpga$ make all
cd fpga && make
make[1]: Entering directory '/home/ee420/Desktop/verilog-ethernet-master/example/C10LP/fpga/fpga'
make[1]: *** No rule to make target '../lib/eth/rtl/iddr.v', needed by 'fpga.qpf'.  Stop.
make[1]: Leaving directory '/home/ee420/Desktop/verilog-ethernet-master/example/C10LP/fpga/fpga'
make: *** [Makefile:14: fpga] Error 2

Could you teach me how to make this project correctly, thx.

alexforencich, 2020-12-11:
Sounds like a symbolic link is not being followed.  Try checking out a clean copy of the repository with 'git clone'.

dennypig, 2020-12-11:
Hello dear author,
I git this project a step-by-step and try make at the C10LP folder,
but the same problem happened,
No rule to make target '../lib/eth/rtl/iddr.v', needed by 'fpga.qpf'

I find the fpgq.qpf this file up and down ,
how do I generate this file for make

thank you so much.

alexforencich, 2020-12-11:
You don't make the file, the file is here: https://github.com/alexforencich/verilog-ethernet/blob/master/rtl/iddr.v .  For some reason, make thinks the file is missing.  I suspect that there is a missing symlink, or the repo is checked out on a filesystem that does not support symlinks.  Alternatively, you can forgo the makefile and manually add all of the files to quartus.