alexforencich/verilog-ethernet #61: ip header checksum error?

cws11, 2020-12-16:
Hi, Alex.
Thanks to you, I am happy to find the verilog-ethernet code useful.

I have a question.
Please look at the picture.

[image]

The red mark in the upper right shows a checksum error.

While trying to find out what's causing it, AXI-STREAM seems to be passing wrong data.
0x45 is the length part of the IP header.
After that, 0x00 is Explicit Congestion Noti, and the next 2 bytes 0x0034 is the length of IP.
It seems to be sending something wrong.
The first axi stream path is axis_gmii_rx -> eth_axis_rx.
The second axi stream path is m_eth_payload_axis in eth_axis_rx -> s_eth_payload_axis in udp_complete.

Finally, looking at the ip header checksum(red mark), it says 0xab8d. Where 8d is the first byte of the my IP address 141.
So it seems that it is an incorrect header checksum.

I am not sure if there is something wrong with the connection.

Thank you.

alexforencich, 2020-12-16:
Something seems really screwy here.  I'm not entirely sure what I'm looking at, but it looks like that's a 32 bit AXI stream interface where only the lowest byte is used, and the rest are simply set to 0?  Not much I can say without seeing how you've hooked things up.

cws11, 2020-12-16:
@alexforencich 
Oh, it is fixed at 32bit datawidth in ILA, but the actual use is 8bit (1byte). I'm newbie to vivado, so I still don't know how to set it up in ILA.
The data width of axi stream is 1 byte.
What I want to ask is that 2bytes of 0x00 0x00 should come after 0x45, but only 0x00 1byte.
Therefore, it seems that the first byte of the ip address was entered as part of the checksum in the process of ip header parsing.
Since I am a software developer, it can be difficult to ask exactly the question.

Thank you.

alexforencich, 2020-12-16:
Ah I see.  Wasn't sure if I was looking at a simulation, ILA capture, or something else.  If it's an ILA capture, then you should be able to export a VCD file that I can examine, instead of a screenshot.  The duplicated byte is not necessarily a problem - there could be a stall in the transfer (tready or tvalid low), but the trace doesn't seem to show that.  If you can send me a VCD trace, then I can try recreating the issue in simulation.

cws11, 2020-12-16:
@alexforencich 

I cannot upload the VCD file, so I uploaded it as a zip file.
[iladata.zip](https://github.com/alexforencich/verilog-ethernet/files/5699930/iladata.zip)

Thank you.

alexforencich, 2020-12-16:
Thanks!  For net_slot_0_... I do not see a tready signal.  Presumably that "extra" byte corresponds to tready falling for one cycle.  On net_slot_1, you can see that only one 0x45 is transferred, as tready is low for the first cycle.  What is the pulse that you say indicates a checksum error?  Is that probe1_1[5]?  What is that connected to in the design?

alexforencich, 2020-12-16:
Yeah one of those zeros is getting lost.  But since I don't see the tready signal, is it possible that it's not connected at all in your design, and that's the cause of the problem?

cws11, 2020-12-16:
@alexforencich 
In case of slot 0, there is no tready signal. Because I have connected the axis_gmii_rx module (not included tready signal) and the eth_axis_rx module.
probe1_1[0] is ip_rx_busy,  probe1_1[5] is ip_rx_error_invalid_checksum signal.
The rest are not particularly important.

alexforencich, 2020-12-16:
Ah, that explains why it isn't working.  You need a FIFO between the output of the MAC and the input of the stack to handle backpressure from the stack without corrupting frames like this.

cws11, 2020-12-16:
@alexforencich 
I didn't know I had to use FIFO.
After applying the FIFO, I will leave the result here.

Thank you very much.

cws11, 2021-01-25:
@alexforencich 
Finally, I can get UDP data using FIFO!

I did a lot of hard work for about a month.
There was no communication between the eth_axis_rx module and the ip_eth_rx module.
I spent quite a long time analyzing the cause.
I guessed the module was stopped due to an ARP transport problem.
After adding the eth_axis_tx module in the UDP complete module, I connected it to gmii and it worked.
Is there any way to completely ignore the ARP request?

Perhaps UDP tx is expected to be simpler.

Thank you very much!

closed 2021-01-26 by cws11