alexforencich/verilog-ethernet #63: UDP Client

dgarigali, 2021-01-18:
Hi!

I am developing a project where I need two independent 10Gb ethernet interfaces: one for only receiving data and another one for transmitting data, both using the UDP protocol.

I have successfully used your examples where the FPGA is the UDP server and UDP is the external client. This is the case where the FPGA only receives data.

Now, I want to use the other interface so that the FPGA is the UDP client and the external PC is the UDP server. Is that possible using the udp_complete source code? 

I already changed it so it is not in loopback. I tried to use the TX AXI-Stream interface to send UDP messages from the FPGA to the PC and it is shown in WhireShark, however, the C Server code running on the external PC does not receive anything. Am I missing something?

alexforencich, 2021-02-20:
It should be possible to use the stack in that way.  If you see the packets in wireshark and they look correct there, then it would seem that the FPGA portion is working correctly.

dgarigali, 2021-03-06:
@alexforencich Thank you for your response.

When the FPGA is the server and the C application is the client, the C client sends an ARP request, then the FPGA sends an ARP response and then the UDP message is actually sent to the FPGA. 

However, when the FPGA is the client, it does not send initially a ARP request to the C server. Is this correct?

Thank you again for your time and support :)

dgarigali, 2021-03-12:
I am closing this because the problem was the ports being used (30 by default, it is advised to use higher number ports) and more importantly the Firewall.

closed 2021-03-12 by dgarigali