alexforencich/verilog-ethernet #65: Broadcom® B50610 PHY chip support

pwang7, 2021-01-21:
Hello,

I'm using the Broadcom® B50610 triple-speed PHY chip, and I'd like the FPGA run as MAC controller to communicate with the PHY chip.

I was wondering whether verilog-ethernet supports this PHY chip or not?
If so, how to run it?

Thanks!

alexforencich, 2021-01-21:
That looks like an RGMII PHY.  In which case the question is probably more about the FPGA than the PHY chip.  What FPGA are you using?  If there is an RGMII design for that FPGA, then it will probably work just fine.  Otherwise, some tweaks may be necessary to some of the DDR interface components so that RGMII will work correctly on that FPGA.

pwang7, 2021-01-21:
Thanks for your reply! @alexforencich 

I'm using ZYNQ 7000, xc7z020clg400-2.

BTW, if I'd like to reuse your RGMII Verilog code, which file to start with?

alexforencich, 2021-01-21:
Take a look at the Nexys Video design, that board uses an RGMII PHY with an Artix 7 FPGA, so it should use the same device primitives as the Zynq 7000.