alexforencich/verilog-ethernet #66: FCS errors on output port

Asm1r, 2021-02-15:
Hi,

I'm trying to create network tap with aggregation support, your code was great help for me since I recently started working with FPGA and even though I don't understand a lot of concepts I was able to nearly finish named project which looks like this:

[image]

On the picture there are four RGMII ethernet phys, 0 and 1 are in the passthrough mode and other two (2 and 3) receive aggregated traffic from 0 and 1. I don't use clk90 since it doesn't work on my TI phys.
The problem I have is that on the ports 2 and 3 (while port 0 and 1 utilization is < 50%) I get FCS errors.
Port 2 behaves a lot better than port 3 even though they are connected to the same broadcast module.
I've also noticed that FCS errors increase a lot when I vary ethernet frame sizes in contrast to fixed size frames.

The board is a custom one with Spartan 7 chip and I've tested each port which works without FCS errors on nearly 100% utilization.
I've also included timing constraints provided for all used modules and timing analysis passes.

I'm really stuck here, I've tried adding more buffers, adding axi register, increasing logic clock speed and nothing worked.
So I'm wondering is this repository meant for my usecase, and if yes, which is the best way to debug this issue?

Thank you, Asmir

alexforencich, 2021-02-17:
RGMII can be a bit finicky to get the timing figured out correctly.  I have never been able to get it to work without clk90; my guess is maybe you're right on the edge of the setup+hold time as required by the PHY chip.  Not sure why they all work fine in isolation but break when used together; that's pretty strange.  Maybe what you need to do is add ODELAY primitives and play around with the tap settings a bit.

Asm1r, 2021-02-23:
Hi Alex, thanks for your response.
I've tried to add odelay to RGMII pins but it turns that you can only route odelay to HP pins and there are none on my Spartan chip.
But I managed to delay the clock by using clk90 input on eth_mac_1g_rgmii  and testing with multiple phase shift (about 5 degrees resolution) - didn't work either.

Turns out I made mistake with inter frame gap, where I put 8 clock cycles instead of standard 12. Situation is a lot better now.

Ports 2 and 3 are working now with maximal 48/48 percent utilizatio on ports 0 and 1, good enough.
Only port 0 now gets FCS errors after a couple of minutes, I'll try to iron out that one too.

Thanks for this great library, Asmir.