alexforencich/verilog-ethernet #68: Query : runt Packets

kewlcreatures, 2021-02-20:
Hi Alex,
Wanted to check a few things with regards to the MAC around a specific topic with regards to runt packets.
1. Is there any possibility that the MAC could generate runt packets under any scenario.
2. Will the MAC ensure to not send out runt packets?

Kind Regards,
Vikas L

alexforencich, 2021-02-20:
If you are not using it with a frame FIFO and the packet is interrupted, then the MAC will terminate the frame with error control characters at any length.  Not sure if you would call that a runt frame or not as the MAC on the other end should drop it as an invalid frame.  Otherwise, it should zero pad the frame out to the specified minimum length (`ENABLE_PADDING = 1`, `MIN_FRAME_LENGTH = 64`).  You can disable that though, in which case you can send runt frames if you want to.  The specified length includes the FCS, so a setting of 64 corresponds to 60 bytes for the header and payload + the 4 byte FCS.

kewlcreatures, 2021-02-22:
Thanks Alex . So we are using the MAC in Frame FIFO Mode.I believe we want to avoid sending a runt so we will have ENABLE_PADDING =1 and MIN_FRAME_LENGTH = 64.

So there's another scenario we've faced. We connect the 10 G cable to the switch and power up the server which powers up the FPGA. In this scenario it seems like the FPGA and possibly the MAC sends link flap errors . This is OK when we are behind an L3 Switch , but when your behind an L1 switch depending on the way the uplink switch is configured it shuts down the port.
Would you have an idea about what's going on in this scenario?

alexforencich, 2021-02-22:
What FPGA, what FPGA board, what switch, and are you using a DAC or some sort of optical cable?  It's entirely possible that my MAC and/or PHY is not doing something that the switch is expecting, unfortunately I'm not sure how hard that's going to be to sort out.

kewlcreatures, 2021-02-22:
thanks for getting back.
so the FPGA board is a U200 and the switch is an L1 Switch ( it just forwards what it gets upstream). I don't know what the upstream switch is and probably we will not get to know that.
So its like:
U200 <--> Our L1 Switch <---> Uplink Switch.

Our L1 Switch just forwards what it gets to Uplink Switch. The uplink switch shows link flap errors when the server is powered up which is when the FPGA is also powered up.

alexforencich, 2021-02-22:
Well, it's kinda hard for me to reason about anything without knowing exactly what the setup is.  However, it's interesting that this seems to be an issue only during boot.  Presumably the FPGA is loading a configuration from flash?  Can you configure the design to hold the Ethernet interface components (transceiver, PHY, MAC, etc.) in reset until software releases it from reset after the OS has booted up?

kewlcreatures, 2021-02-22:
Thanks a lot for your inputs. Indeed they are very helpfull. Yes we are probably contemplating doing the same holding up the components in reset untill the software releases it from reset after the OS has booted up..