alexforencich/verilog-ethernet #70: Netcat doing nothing

surelynottrue, 2021-03-11:
Hi!
So we were trying to implement your example on the alterra DE2-115 EP4CE115F29C7N board. The issue we are facing here, is that everything compiles nicely, the displays light up, switches work, but `netcat` does nothing. We tried disabling DHCP, setting a manual IP address on the board, different from 128, but `netcat -u 192.168.1.128 1234` just does nothing. We even tried connecting the wire directly and through a router.
Is there something that we are missing somewhere? Any help will be appreciated, thanks!

alexforencich, 2021-03-11:
What do you see with tcpdump?

surelynottrue, 2021-03-11:
Nothing apart from a few what look like whois requests from the host

alexforencich, 2021-03-11:
so the host is sending ARP requests for 192.168.1.128, but there is no response from the board?

surelynottrue, 2021-03-11:
We do not see any mention of 192.168.1.128. There are some requests to a samba share on our raspberry pi, just attempts. Netcat doesnt give out any requests. It doesnt stay on, like it should.

alexforencich, 2021-03-11:
well, that indicates to me that there is a network configuration issue on the host.  Presumably the host has ip 192.168.1.xxx with subnet mask 255.255.255.0?

surelynottrue, 2021-03-12:
It seems it was a network configuration issue afterall. The subnet mask wasn't properly set up, leading to all the problems we were having. Thank you for your time!

surelynottrue, 2021-03-14:
Hi, so I am bothering you again. Some things are happening that do not make sense to me. This used to work, doesnt anymore.
1. I have manually set my Mac's IP to 192.168.1.199 with subnet mask 255.255.255.0, nothing on the router field. The fpga has been connected directly, using the thunderbolt port of the 2017 Air.
2. Here's my configuration:
       `en3: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	options=50b<RXCSUM,TXCSUM,VLAN_HWTAGGING,AV,CHANNEL_IO>
	ether a8:60:b6:05:b0:6e
	inet6 fe80::df:fa33:51ad:aa0%en3 prefixlen 64 secured scopeid 0x4
	inet 192.168.1.199 netmask 0xffffff00 broadcast 192.168.1.255
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect (1000baseT <full-duplex,flow-control>)
	status: active`
    WiFi turned off, while testing.
3. When I run `nc -u 192.168.1.128 1234`, and send a hello, I see the light blink on the RX LED on the fpga, but that is all that it does. Nothing comes back to me, and the 7segment display doesnt change to the IP address of y computer.

Please let me know if you need any more information. I have programmed the same fpga with the corresponding example project using Quartus Prime version 20.1.1 Build 720 11/11/2020 SJ Lite Edition

alexforencich, 2021-03-14:
What does TCP dump show?

surelynottrue, 2021-03-14:
I tried monitoring the connection over wireshark, and all that was detected was ARP requests. Nothing else at all. Would you like me to do any troubleshooting step?

alexforencich, 2021-03-14:
Well, do the ARP requests look correct?  Also, the DE2-115 has two Ethernet ports, but only one of them is connected to the UDP stack in the example design.  Are you sure you're using the correct port?

zaidtahirbutt, 2022-06-22:
> well, that indicates to me that there is a network configuration issue on the host. Presumably the host has ip 192.168.1.xxx with subnet mask 255.255.255.0?

Thanks for this suggestion @alexforencich. I don't have much knowledge of networking so this comment helped to establish connection using the netcat command (set my host ethernet ip to 192.168.1.200 with subnet mask 255.255.255.0).

myqlee, 2022-10-21:
> 

IP must be 192.168.1.xxx? Can it be set to 192.168.30.xxx? @zaidtahirbutt

myqlee, 2022-10-21:
> 

> Thanks for this suggestion @alexforencich. I don't have much knowledge of networking so this comment helped to establish connection using the netcat command (set my host ethernet ip to 192.168.1.200 [...]

IP must be 192.168.1.xxx? Can it be set to 192.168.30.xxx? @zaidtahirbutt

zaidtahirbutt, 2022-10-21:
> IP must be 192.168.1.xxx? Can it be set to 192.168.30.xxx? @zaidtahirbutt

@myqlee the network part of the host and the fpga board must be the same, i.e., if the host is set to 192.168.30.xxx, then the fpga board should have the same network address 192.168.30.xxx, considering the subnet mask to be 255.255.255.0.