alexforencich/verilog-ethernet #71: UDP Fragmentation

Uroojt, 2021-03-22:
I have to send 50,000 bytes per second on UDP protocol, as per standard I can only send 1500 bytes in one UDP packet, so I have to break my data into chunks of 1500 bytes(fragments), but when I communicate over network code choke after certain number of packets, and sometimes if it doesn't choke the receiving packets on Wireshark are broken into some misaligned packets. Is there anything that you can help with?

Beside that when I try to open multiple udp ports on same ip address this code doesn't accept the reception packet from latter port once it has established the connection with former reception port. any help?

alexforencich, 2021-03-22:
Fragmentation is not supported, for either TX or RX.  Have you tried jumbo frames?  It should be possible to set the MTU higher than 1500 bytes in most systems.  

If you're getting corrupted packets, send me a wireshark trace or something and I can take a look.  

The current stack is stateless and basically just does parsing and de-parsing, aside from the implementation of the ARP layer.  So I'm not sure what you mean by not accepting packets, it should accept all well-formed UDP packets.

Uroojt, 2021-03-22:
Many Thanks for your quick response.
Does this code support jumbo frame?
I'll share the Wireshark trace in a while.

for multiple UDP port,
if send packet on 1234 it get received with this code, but on the same session if I send packet on udp port 1236 using same ip address, it doesn't get accepted by this code(data is not received), or rx source port doesn't get updated to 1236 .

Is this design full-duplex? When using it for 2 ethernet PHYs, what will be the dependency of ready signals of 2 PHYs on eachother ?

alexforencich, 2021-03-22:
The example design is just a guide line, you may need to change the extremely basic filtering code to match the port that you want to use.  Another option is perhaps to open multiple connections with the same destination port but different source ports.  

If you want to use it with two Ethernet PHYs, then presumably you would use two different instances of the stack, one for each PHY.  That way, each PHY gets its own MAC, MAC address, IP address, etc.

Uroojt, 2021-03-22:
Does this code support Jumbo frames?

alexforencich, 2021-03-22:
I think it does, no part of the code should limit the frame size unless there is a bug somewhere.

Uroojt, 2021-03-22:
Many Thanks.
I have implemented 2 separate stacks for 2 different PHY's, Is there any relationship between tx ready of both instances?, like if I want to send that from one PC to PHY1 and then from PHY1 to PHY 2 and from PHY2 to PC two, I'm unable to cater the ready signal. is there any timing constrains or timing diagram apart from the one available  in read me file.

alexforencich, 2021-03-22:
I mean, the PHY doesn't have a ready signal on the RX side, so at some point you just have to start dropping frames.

Uroojt, 2021-03-22:
Thanks allot for your prompt response and help

closed 2021-03-22 by Uroojt