alexforencich/verilog-ethernet #72: question about   principle

AliceXuXu, 2021-04-03:
Is KC705 full duplex? Because the light of the duplex of my development board is on

AliceXuXu, 2021-04-04:
please help me  ,if someone know it.thank you

alexforencich, 2021-04-04:
The PHY on that board is wired up exactly the same as on several other boards; it should come up as full duplex if the link partner supports full duplex.  The verilog-ethernet example design does not touch any of the PHY registers, and the default for autonegotiation should be to request full duplex.  I'll dig out my KC705 later and see if the duplex light is on or not just to confirm that it's working correctly.

AliceXuXu, 2021-04-05:
Thanks for your reply.here I would show you two pictures.the first is Xilinx offical document:url:https://www.xilinx.com/support/documentation/boards_and_kits/kc705/ug810_KC705_Eval_Bd.pdf
the second is the result of my experiment
[image]

AliceXuXu, 2021-04-05:
[image]

AliceXuXu, 2021-04-16:
> The PHY on that board is wired up exactly the same as on several other boards; it should come up as full duplex if the link partner supports full duplex. The verilog-ethernet example design does not [...]

Have you install bitstream in you KC705 board？，If you reply to me, it will be a great honor

alexforencich, 2021-04-30:
Finally got around to testing this.  Looks like the light is simply inverted from what the documentation states.  I connected my KC705 to a gigabit USB NIC.  ethtool reported that the link came up as 1000 Mbps, full duplex.  The lights 1000 and DU were illuminated on the card.  Then I used ethtool to set the speed to 100 Mbps, and the lights changed to 100 and DUP.  Then I used ethtool to set the duplex to half, and the DUP light went out, leaving only the 100 light illuminated.

AliceXuXu, 2021-05-01:
OK, thank you very much. That means the official document is wrong. Thank you very much again. I was wondering if you could offer some suggestions on encapsulating UDP frames. After all, this project is just testing the Ethernet function. Do you have any ideas for practical applications?