alexforencich/verilog-ethernet #73: Cannot receive packets

abdullahyildiz, 2021-04-06:
I'm trying to use [ariane-ethernet](https://github.com/lowRISC/ariane-ethernet/tree/6a5436bf110f83ebb13119dbd82650ccd8f947c9) design in [OpenPiton+Ariane](https://github.com/PrincetonUniversity/openpiton) project, which is a Linux capable SoC design.

There is no problem while compiling the RTL design and generating the bitstream (timing is met).

The problem is that when I try to run ping command after the system boots, OS cannot receive the ARP response and hence cannot proceed by sending ICMP packets.

I also came across a situation where ping worked but later it again stopped working.

@alexforencich, did you encounter a problem like this before?

You can refer to the original [issue](https://github.com/lowRISC/ariane-ethernet/issues/3) for the details.

Reisswolf, 2021-04-13:
The project currently does not respond to ping request because it does not support ICMP. You could add the support for ICMP by implementing the missing functionality. I suggest to start with the Echo request and Echo reply features. [Wikipedia](https://en.wikipedia.org/wiki/Ping_(networking_utility)) contains most of the information that you would need to get started.

alexforencich, 2021-04-13:
@Reisswolf I think the ariane-ethernet project only uses the MAC, not the whole stack.  

I have not seen this issue before, but it looks like that code uses RGMII.  RGMII can be a bit tricky to get working as it is a DDR interface and requires some additional work to ensure proper setup and hold times around the clock edges.  You may need to add iodelay elements and adjust the delay settings to get it to work reliably.  The required settings will be board-dependent.

abdullahyildiz, 2021-04-13:
@alexforencich, @Reisswolf,

Thanks for your support.

I am using only the MAC interface in my design.

[ariane-ethernet](https://github.com/lowRISC/ariane-ethernet/tree/6a5436bf110f83ebb13119dbd82650ccd8f947c9) is being used in a similar [project](https://github.com/openhwgroup/cva6/tree/master/fpga/src) (with RGMII) and it seems that respective IDELAY values are set to zero in [rgmii_soc.sv](https://github.com/lowRISC/ariane-ethernet/blob/6a5436bf110f83ebb13119dbd82650ccd8f947c9/rgmii_soc.sv).

The constraints also do not include min and max delay [constraints,](https://github.com/openhwgroup/cva6/blob/fbe1e01e2090b19bcf49d9d8c828cceeb7daab12/fpga/constraints/genesys-2.xdc#L60-L67) that is why I wanted to ask this.

myqlee, 2022-10-28:
> @alexforencich, @Reisswolf,
>
> Thanks for your support.
>
> I am using only the MAC interface in my design. [...]

As you said, ariane is a mac, does it function normally now?

myqlee, 2022-10-28:
> @alexforencich, @Reisswolf,
>
> Thanks for your support.
>
> I am using only the MAC interface in my design. [...]

As you said, [ariane-ethernet](https://github.com/lowRISC/ariane-ethernet/tree/6a5436bf110f83ebb13119dbd82650ccd8f947c9) is a mac, does it function normally now?