alexforencich/verilog-ethernet #74: Tight timing in rgmii_phy_if.v

eugene-tarassov, 2021-04-09:
When USE_CLK90 == "TRUE", phy_rgmii_tx_clk register has very tight timing on the data path: clk90 - clk = 2ns. Vivado often fails to meet this timing on Artix-7 FPGA.

I made few changes rgmii_phy_if.v to relax the timing. The idea is to clock TX output with posedge clk and negedge clk90 instead of posedge clk90 and posedge clk. Now the requirement is 6ns.

My version of the file:
[https://github.com/eugene-tarassov/vivado-risc-v/blob/master/ethernet/rgmii_phy_if.v](https://github.com/eugene-tarassov/vivado-risc-v/blob/master/ethernet/rgmii_phy_if.v)
The diff is:
```
$ diff ethernet/rgmii_phy_if.v ethernet/verilog-ethernet/rtl/rgmii_phy_if.v
217c217
<     .clk(clk),
---
>     .clk(USE_CLK90 == "TRUE" ? clk90 : clk),
229c229
<     .clk(USE_CLK90 == "TRUE" ? ~clk90 : clk),
---
>     .clk(clk),

```
Could you consider these changes?

alexforencich, 2021-04-13:
That's certainly an interesting idea.  However, I wonder if the input pins to the DDR flip flop may need to be swapped so that the edges have the correct polarity.  I may need to run some experiments to check this.

eugene-tarassov, 2021-04-13:
I run RISC-V Linux on Nexys Video board: [vivado-risc-v](https://github.com/eugene-tarassov/vivado-risc-v), and use ethtool to try different speeds 10, 100 and 1000, full and half duplex, and everything seems working fine.

alexforencich, 2021-04-13:
Gotcha.  Well, I don't have access to a Nexys Video board, so I'll test it out on probably a KC705 and see what it looks like.  The lower speeds should be no problem, the only possible issue would be with gigabit speed, and even then I think the main issue would be proper handling of the frame error signal, so you may not see any obvious problems under most conditions.