alexforencich/verilog-ethernet #76: how to make it onto U50

fyflxl, 2021-04-20:
well, i 'm a newcomer.i have successfully make the vivado project,but what should i do next to program it onto U50?do i need to make a mcs and then program it?thanks a lot!

alexforencich, 2021-04-20:
You need to use an Alveo programming cable, part number HW-DMB-1-G: https://www.xilinx.com/products/boards-and-kits/alveo/accessories.html

Using that cable, simply load fpga.bit onto the card via JTAG.  

Do NOT use the Alveo card management tools to flash the design to the card.  Since this design doesn't bring up the PCIe interface and expose the flash to the host, you can only do that once, and then you will need the JTAG cable to fix it!  Corundum is a slightly different story as it has its own in-band firmware update mechanism.

fyflxl, 2021-05-12:
and what kind of cable i could use to qsfp28? i have a 40G QSFP+ to 4*10SFP+ cable, but it did not work

alexforencich, 2021-05-12:
QSFP+ to 4x SFP+ should work fine, but you have to make sure to use the first cable (they should all be labelled in some way) as that's the only one that the example design connects to the stack (the other three just have 10G PHYs tied off to send idles continuously, so it should show the link being up, but it won't respond).  Well, presuming your NIC is happy with the breakout cable, some are more picky than others.