alexforencich/verilog-ethernet #79: Question about flow control

Yuan-Mao, 2021-05-25:
Hi,

I was trying to figure out if there's any flow control mechanism in the MAC module.

I was looking at the interface of 'eth_mac_10g' module and seems like there's no signal for the AXI receiver to indicate that it is overwhelmed by some sender, so I guess there's no flow control in this module? Just wonder if I have missed something?

Thank you very much.

alexforencich, 2021-05-25:
How is the MAC supposed to exert flow control against the PHY?  How is the PHY supposed to exert flow control against the serdes?  etc.

Yuan-Mao, 2021-05-25:
To my understanding, the flow control mechanism sits in L2, so I think I'm asking about how the MAC exerts flow control against the PHY in the design?

alexforencich, 2021-05-25:
The MAC cannot exert flow control against anything.  If the downstream logic is not ready for a frame and there is no space in the FIFO, the frame gets dropped.

alexforencich, 2021-05-26:
Also, I should note that pause frames are simply a request that the other end stop sending data, potentially for a specific traffic class.  They require the link partner to honor the request and stop sending, and you have to have enough buffer space to hold whatever packets might currently be in flight when you send the request in the first place, so you can still run in to issues with RX buffer overflow even when pause frames are used.  I don't know of any MAC that implements a tready or equivalent flow control handshake signal on the MAC RX side.  However, many MACs do provide some additional logic for sending and receiving pause frames, which I may add to my MACs at some point.  Note that this logic does nothing in terms of controlling the flow of data, it simply serves to "jump the queue" on both the transmit and receive side and simplify the generation and reception of the pause frames.

Yuan-Mao, 2021-05-26:
I see. Thank you for your detailed explanation. Everything makes a lot of sense now.

closed 2021-05-28 by alexforencich