alexforencich/verilog-ethernet #80: Marvell Alaska 88e1512 support

AyeshaGauhar, 2021-06-03:
Hi 

I am using xc7z020clg400-2 with Marvell Alaska 88e1512 phy chip. I took the nexys video board design as reference. For reception( PC to board) it is working fine. But when transmitting from board to pc, no packet is recieved on PC/Wireshark. Does the code need some modification for this phy ic?

Looking forward to your response

alexforencich, 2021-06-03:
It's a bit tricky to get RGMII working correctly since it is a DDR interface and as such is significantly more sensitive to setup/hold timing vs. other interfaces.  It's also more dependent on board layout.  You may need to play around with output delays and/or the phase offset on the (nominally) 90 degree clock to get things working correctly.

AyeshaGauhar, 2021-06-05:
Many thanks for your response. I'll look in to it

yoursunny, 2022-06-09:
I'm attempting to use this library on [snickerdoodle black + piSmasher baseboard](https://krtkl.com/snickerdoodle/).
It contains:

* Xilinx Zynq-7020, same as the Zedboard
* Marvell 88E1512 PHY
* PHY signals are wired to Zynq PL pins, except that the PHY RESET signal is wired to Zynq PS MIO pin

Given @alexforencich mentions the difficulty in RGMII timing and Xilinx provides a *Gmii to Rgmii* IP for free, I had the idea of using that IP together with a GMII interface generated by this library.
I started with `example/KC705/fpga_gmii` that provides a GMII interface.
It's connected like this:

[image]

Code modifications:

* In `fpga.v`, delete the differential clock, and expose `clk_200mhz_ibufg` as 200MHz clock input.
* Modify `fpga_core.v`: in `eth_mac_inst` instance set `.CLOCK_INPUT_STYLE("BUFG")`.
  Otherwise there would be DRC RTSTAT-1 error regarding `rx_ssio_sdr_inst/clk_io`.

This design can generate bitstream, but it doesn't seem to work on hardware: no response to ARP, same as my previous attempt in https://github.com/alexforencich/verilog-ethernet/issues/112#issuecomment-1149879587 .
I'd appreciate some hints on whether this is a viable direction.