alexforencich/verilog-ethernet #81: porting to VU440

Albert-Siu, 2021-06-07:
I try to port to VU440 but fail to make the project. My steps is as below. Please provide some advice. Thx.

1. Copy from project VCU108/fpga_1g to VU440/fpga_1g/
2. Type 
3. Modify device part number and pin assignment
4. Build the project again

I fail in step 2 which is make the project with the original VCU108/fpga_1g/. The error message is as below:

make
cd fpga && make
make[1]: Entering directory `/home/XXXX/project/fpga/eth/mod/verilog-ethernet-master/example/VU440/fpga_1g/fpga'
make[1]: *** No rule to make target `fpga.bit', needed by `fpga'.  Stop.
make[1]: Leaving directory `/home/XXXX/project/fpga/eth/mod/verilog-ethernet-master/example/VU440/fpga_1g/fpga'
make: *** [fpga] Error 2

alexforencich, 2021-06-07:
Sounds like a missing file or broken symlink.  Does the VCU108 design build?  Does your copy build, before making any modifications?

Albert-Siu, 2021-06-07:
Yes. There are some symlink broken. I can create the project with the makefile for VCU108.

There are 3 errors in syn process. 

ERROR: [Common 17-55] 'get_property' expects at least one object. [/home/XXXX/project/fpga/eth/mod/verilog-ethernet/example/VU440/fpga_1g/lib/eth/syn/vivado/eth_mac_fifo.tcl:23]

ERROR: [Common 17-55] 'get_property' expects at least one object. [/home/XXXX/project/fpga/eth/mod/verilog-ethernet/example/VU440/fpga_1g/lib/eth/lib/axis/syn/vivado/axis_async_fifo.tcl:23]

ERROR: [Project 1-581] Command stopped due to earlier errors.