alexforencich/verilog-ethernet #82: no axis input while using rgmii_1g_fifo on Spartan-7

Winters123, 2021-06-14:
Hi Alex,

I used you `rgmii_1g_fifo` for one of our open-source projects running on spartan-7 (part num: `xc7s50fgga484-1`).  While the logic can be synth & impl on the board, it cannot get any data from the axis side of the MAC. We cannot figure out what could possibly go wrong. 

We followed the doc to connect the pins to PHY as this:

```
set_property PACKAGE_PIN G21 [get_ports {rgmii_rxd[3]}]
set_property PACKAGE_PIN F22 [get_ports {rgmii_rxd[2]}]
set_property PACKAGE_PIN F21 [get_ports {rgmii_rxd[1]}]
set_property PACKAGE_PIN E22 [get_ports {rgmii_rxd[0]}]
set_property PACKAGE_PIN E18 [get_ports {rgmii_txd[3]}]
set_property PACKAGE_PIN E17 [get_ports {rgmii_txd[2]}]
set_property PACKAGE_PIN D19 [get_ports {rgmii_txd[1]}]
set_property PACKAGE_PIN E19 [get_ports {rgmii_txd[0]}]
set_property PACKAGE_PIN E16 [get_ports e_mdc]
set_property PACKAGE_PIN G14 [get_ports e_mdio]

create_clock -period 8 [get_ports rgmii_rxc]
# means 125MHz for rx clock. 
###
set_property PACKAGE_PIN G22 [get_ports rgmii_rxc]
set_property PACKAGE_PIN D22 [get_ports rgmii_rxctl]
set_property PACKAGE_PIN F19 [get_ports rgmii_txc]
set_property PACKAGE_PIN F20 [get_ports rgmii_txctl]
```

And in the `top.v`, the mac is connected like this:
```
eth_mac_1g_rgmii_fifo #
(
    // target ("SIM", "GENERIC", "XILINX", "ALTERA")
    .TARGET("XILINX"),
    // IODDR style ("IODDR", "IODDR2")
    // Use IODDR for Virtex-4, Virtex-5, Virtex-6, 7 Series, Ultrascale
    // Use IODDR2 for Spartan-6
    .IODDR_STYLE("IODDR"),
    // Clock input style ("BUFG", "BUFR", "BUFIO", "BUFIO2")
    // Use BUFR for Virtex-5, Virtex-6, 7-series
    // Use BUFG for Ultrascale
    // Use BUFIO2 for Spartan-6
    .CLOCK_INPUT_STYLE("BUFG"),
    // Use 90 degree clock for RGMII transmit ("TRUE", "FALSE")
    .USE_CLK90("FALSE"),
    .AXIS_DATA_WIDTH(AXIS_DATA_WIDTH),
    // .AXIS_KEEP_ENABLE(),
    .AXIS_KEEP_WIDTH (AXIS_KEEP_WIDTH),
    .ENABLE_PADDING(1),
    .MIN_FRAME_LENGTH(64),
    // .TX_FIFO_DEPTH(),
    .TX_FIFO_PIPELINE_OUTPUT(2),
    .TX_FRAME_FIFO(1),
    // .TX_DROP_BAD_FRAME(),
    .TX_DROP_WHEN_FULL(0),
    // .RX_FIFO_DEPTH(),
    // .RX_FIFO_PIPELINE_OUTPUT(),
    .RX_FRAME_FIFO(1)
    // .RX_DROP_BAD_FRAME(),
    // .RX_DROP_WHEN_FULL()
)rgmii_mac_1g
(
    .gtx_clk(rgmii_rxc),  //use 125Mhz    //use received 125M clk
    .gtx_clk90(rgmii_rxc),              // did not using 90degreee clock
    .gtx_rst(~i_sys_rst_n),
    .logic_clk(i_sys_clk_IBUFG),      // system 50MHz system clk
    .logic_rst(~i_sys_rst_n),

    /*
     * AXI input
     */
    .tx_axis_tdata(tx_axis_tdata),
    .tx_axis_tkeep(tx_axis_tkeep),
    .tx_axis_tvalid(tx_axis_tvalid),
    .tx_axis_tready(tx_axis_tready),
    .tx_axis_tlast(tx_axis_tlast),
    .tx_axis_tuser(tx_axis_tuser),

    /*
     * AXI output
     */
    .rx_axis_tdata(rx_axis_tdata),
    .rx_axis_tkeep(rx_axis_tkeep),
    .rx_axis_tvalid(rx_axis_tvalid),
    .rx_axis_tready(rx_axis_tready),
    .rx_axis_tlast(rx_axis_tlast),
    .rx_axis_tuser(rx_axis_tuser),

    /*
     * RGMII interface
     */
    .rgmii_rx_clk(rgmii_rxc),
    .rgmii_rxd(rgmii_rxd),
    .rgmii_rx_ctl(rgmii_rxctl),
    .rgmii_tx_clk(rgmii_txc),
    .rgmii_txd(rgmii_txd),
    .rgmii_tx_ctl(rgmii_txctl),

    /*
     * Status
     */
    .tx_error_underflow(),
```

Could you please have a look at it and give us some hints if possible?  

Thank you very much!

alexforencich, 2021-06-14:
RGMII is a DDR interface, which makes things a bit tricky.  I have never gotten it to work before without the 90 degree clock, and if you don't use the 90 degree clock you'll probably need to do quite a bit of fiddling with IODELAY primitives.  You may need to fiddle with IODELAY primitives anyway to get the RX side working.  I recommend probing the internal GMII interface between the MAC and the PHY IF with an ILA to see what's going on there; that usually sheds some light onto what's going on.

alexforencich, 2021-06-15:
Well, that makes no sense.  If rx_dv falls, then tvalid should fall after a clock cycle or two.  Is that tvalid output before or after the RX FIFO?

zhuanshulz, 2021-06-16:
thank alex a lot!
we finally found that the **rx_axis_tready** input signal does not connected to the corresponding wire which cause the waiting of the axi-lite bus.

closed 2021-06-17 by Winters123