alexforencich/verilog-ethernet #89: UDP complete waves

aignacio, 2021-07-31:
Hey Alex,

tks again for the IPs, bothering you again about usage! do you know how to get some waves of `udp_complete` module or if there's a diagram of how to drive the interfaces on this module?  I tried to take a look at the testbench but isn't clear how it works with myhdl exactly...

alexforencich, 2021-07-31:
My recommendation is to take a look at the example design for how to hook everything up.  

The testbench has not been converted to cocotb yet (I am doing that piecemeal as I rewrite modules in a parametrized way) but you can run the myhdl testbench, and it will dump out an lxt waveform dump.  You'll just have to install the myhdl python package, and compile and install myhdl.vpi so that icarus verilog can find it.  Then you can run the testbench.

aignacio, 2021-07-31:
Thanks for the hint, is that possible to use only s/m_udp_* I/F if I only want to send UDP pkts?

aignacio, 2021-07-31:
[image]
 got the waves as you suggested, tks!, on this test I'm seeing you're setting 40 bytes as the length but only 32 bytes are transferred, is that fine?

alexforencich, 2021-07-31:
The UDP length field includes the header, which is 8 bytes.  However, if UDP checksum generation is enabled, then you can tie that off to zero and the length fields will be computed from the actual length of the AXI stream payload data.

aignacio, 2021-08-02:
In the `test_udp_complete.py` I see that the parameter is set (UDP_CHECKSUM_GEN_ENABLE == 1) and you send the checksum and subtract it from the total (40). You mean if this parameter is set to ZERO I can tie off (s_udp_checksum = '0) and consider length == total number of data streamed in axis?

alexforencich, 2021-08-02:
No, if checksum generation is enabled (should be enabled by default), then the length and checksum fields are ignored.  If you disable checksum generation, then you must specify the length and checksum correctly for the packet to be valid.

unbtorsten, 2021-10-05:
First of all, thank you for this amazing core! I, too, struggle with the documentation. Hence my question that relates to the OP:

> My recommendation is to take a look at the example design for how to hook everything up.

I have been working with the examples (cf. #93) and have modified the loopback implemented in fpga_core.v. Furthermore, I am able to send out custom data by diverting the rx_fifo stream and feeding the udp_complete instance from a second tx_fifo.

Is this how you designed integration of the module into a project? I feel like I do not understand the difference between fpga.v and fpga_core.v. In fpga.v, the Ethernet core offers an (unused) UART. Should I connect to the core at this level, or instantiate other modules (such as data generator and processing) in fpga_core.v?

Juan8UTN, 2022-10-07:
> I have been successful in working with the examples (cf. #93) and have modified the loopback implemented in fpga_core.v. Furthermore, I am able to send out custom data by diverting the rx_fifo stream and feeding the udp_complete instance from a second tx_fifo.

Hi, sorry for bother you, but i'm asking myself if you can help me with that. I need to send custom data trough ethernet using UDP and, because i'm a newbie in the field, i'm little lost with that. Could you help me with some tips, code, whatever you consider?

unbtorsten, 2022-10-24:
> Hi, sorry for bother you, but i'm asking myself if you can help me with that. I need to send custom data trough ethernet using UDP and, because i'm a newbie in the field, i'm little lost with that. Could you help me with some tips, code, whatever you consider?

I think your question is rather broad at this time. This makes it difficult to help. Can you be more specific with regard to what you are aiming to achieve and what you have tried already? Also, this may better be posted in a separate issue?

Juan8UTN, 2022-10-25:
> I think your question is rather broad at this time. This makes it difficult to help. Can you be more specific with regard to what you are aiming to achieve and what you have tried already? Also, [...]

Hi, first of all, thanks for take the time to respond to me, i have already posted a separate issue (issue #133). Briefly, i need to modified the loopback example to respond with some custom data. I tried to modify the instances udp_complete and udp_payload_fifo but i hadn't any success. i am really newbie in this field, and i find myself a little lost with this, and really need it to continue with my final project to obtain my Bacherlor's degree.